basler/pypylon #404: _genicam.TimeoutException: Write operation failed

jainvi-wwt, 2021-08-23:
We are using a camera with arm64 architecture and model name **Basler daA1600-60ucADL**

We are trying to use software trigger to capture images at regular intervals. Below is the function run() that we call using a multiprocessing framework to capture frames. The code works fine for a certain amount of time (~50-60 hrs) after which it throws a timeout exception shown below:

**Error**:
```
 八  21 22:14:38 adlink-desktop main.py[8408]: Traceback (most recent call last):
 八  21 22:14:38 adlink-desktop main.py[8408]:   File "/home/adlink/production_git/src/Camera.py", line 56, in run
 八  21 22:14:38 adlink-desktop main.py[8408]:     camera.ExecuteSoftwareTrigger()
 八  21 22:14:38 adlink-desktop main.py[8408]:   File "/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py", line 3957, in ExecuteSoftwareTrigger
 八  21 22:14:38 adlink-desktop main.py[8408]:     return _pylon.InstantCamera_ExecuteSoftwareTrigger(self)
 八  21 22:14:38 adlink-desktop main.py[8408]: _genicam.TimeoutException: Write operation failed on device '2676:ba03:2:1:2' at address: 0x10000048; length 4; Error: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsbDevice.cpp', line 827)
```

**Code snippet**:

```
def run(main_q,cfg):
    camera = None
    frame_diff = []
    try:
        tl_factory = pylon.TlFactory.GetInstance()
        camera = pylon.InstantCamera()
        camera.Attach(tl_factory.CreateFirstDevice())
        camera.OutputQueueSize = 1
        camera.MaxNumBuffer = 3000
        camera.Open()
        pylon.FeaturePersistence.Load('low_res.pfs'), camera.GetNodeMap(), True)
        camera.Close()
        time.sleep(1)
        camera.RegisterConfiguration(
            pylon.SoftwareTriggerConfiguration(),
            pylon.RegistrationMode_ReplaceAll,
            pylon.Cleanup_Delete)
        image_event_handler = None
        if str(cfg['Camera']['type']) == 'video_save':
            image_event_handler = Camera_trigger.Queue_write_handler(cfg,main_q)

        sleep_time = 0.030

        camera.RegisterImageEventHandler(image_event_handler,
                                         pylon.RegistrationMode_Append,
                                         pylon.Cleanup_Delete)

        camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)
        grab_pair_frames_per_sec = 4
        time_elapsed_per_sec = 0

        while True:
            camera.ExecuteSoftwareTrigger()
            time.sleep(sleep_time)
            camera.ExecuteSoftwareTrigger()
            time.sleep(sleep_time)
    except genicam.GenericException as e:
        logger.exception("An exception occured.", e)
        camera.Close()
    except KeyboardInterrupt:
        camera.Close()
    finally:
        logger.info('closing program after exception')
        camera.Close()
```

SMA2016a, 2021-08-24:
it sounds like if the usb host controller is not in good shape. If an error has occurred, what action is needed to get the system in good shape? 
is the camera accessible in pylon viewer after this error?

achal-wwt, 2021-08-25:
@SMA2016a, We didn't check accessing the camera using the pylon viewer after the error. We can try that next time when we face the error.

jainvi-wwt, 2021-09-10:
@SMA2016a we have confirmed that the code can be restarted automatically (using a service) any time it gets stopped due to this exception, meaning we do not need to reboot the camera to get the code running again.

closed 2025-12-15 by thiesmoeller