basler/pypylon #407: No longer able to see device in Pylon IP Configurator after failure to set static IP

yuto-shinagawa, 2021-08-29:
I was attempting to use the Pylon IP Configurator (Linux) to assign a static IP to my Basler acA1440-73gm GigE camera.
I have both my laptop and camera connected via a Gigabit rated network cable to a PoE switch.  I'm using an ethernet to USB adapter on my laptop to connect to the switch.

I happened to take a screenshot (for purposes of documenting) right before I hit save:

[image]

Upon saving, a window popped up, asking me to choose a temporary IP address for it to use during this process.
The process appeared to take a while, during which the mouse cursor was moving to indicate that it was waiting for the application.  After what seemed to be about 60 seconds, I got a terse error to the effect of "Failed to set configuration."  Exact wording likely differs.  But thereafter, I am no longer able to see the device in the IP Configurator.  I noticed the camera module was very hot, but this may be expected and unrelated to this issue.  The power LED light is solid and the Gigabit link LED is flashing, as expected on the network switch port to which the camera connects.

Are the procedures in #380 what I need to follow?  Are there alternative fixes to this?

SMA2016a, 2021-09-03:
first of all, please refer to the pylon's README. disable the IP filter.

if that does not help, try following.

In order to re-configure a camera by using its MAC address, run the IP Configuration tool and press the following keys simultaneously:
 
Ctrl + Shift + Alt + F

Now enter the camera MAC address separated by hyphen (-) and preferrably use the suggested IP Address, Subnet Mask and Default Gateway (on Windows) or enter any persistent IP address of Class C (or an address within the subnet of the network adapter the camera was connected to) if you were using Linux, e.g.:

    IP Address: 192.168.3.2
    Subnet Mask: 255.255.255.0
    Default Gateway: 0.0.0.0

yuto-shinagawa, 2021-09-14:
Thank you kindly.  It turns out we didn't need to disable the IP filter nor reconfigure the camera using its MAC address.

We succeeded in connecting to the camera and assigning an IP when we reattempted my original procedures on a different computer with a dedicated Ethernet port.  I suspect the USB-RJ45 dongle that we used the first time around was the culprit.

Closing issue.

closed 2021-09-14 by yuto-shinagawa