basler/pypylon #408: OnCameraDeviceRemoved event received after system being idle for some time

jsiebert, 2021-09-08:
Hey there,

I'm running a system with multiple GigE Cameras which grabs images using a hardware trigger (only one frame per trigger). The system is required to run 24/7 but there might be some time when it is idle, for example during the night. The system runs on Ubuntu 20.04 LTS with Python 3.8.10 and PyPylon 1.7.2.

Now, when I run my code and leave the system without triggering the hardware trigger for some time, it receives an OnCameraDeviceRemoved event as one can see from this log (last two entries):
```
2021-09-07 13:06:10,504 OnAttach event
2021-09-07 13:06:10,504 OnAttached event for device: acA3088-16gm
2021-09-07 13:06:10,504 OnOpen event for device: acA3088-16gm
2021-09-07 13:06:10,540 OnOpened event for device: acA3088-16gm
2021-09-07 13:06:12,767 OnAttach event
2021-09-07 13:06:12,767 OnAttached event for device: acA3088-16gm
2021-09-07 13:06:12,768 OnOpen event for device: acA3088-16gm
2021-09-07 13:06:12,801 OnOpened event for device: acA3088-16gm
2021-09-07 13:06:15,012 OnAttach event
2021-09-07 13:06:15,012 OnAttached event for device: acA3088-16gm
2021-09-07 13:06:15,012 OnOpen event for device: acA3088-16gm
2021-09-07 13:06:15,046 OnOpened event for device: acA3088-16gm
2021-09-07 13:06:17,261 OnAttach event
2021-09-07 13:06:17,262 OnAttached event for device: acA3088-16gc
2021-09-07 13:06:17,262 OnOpen event for device: acA3088-16gc
2021-09-07 13:06:17,301 OnOpened event for device: acA3088-16gc
2021-09-07 13:06:19,996 OnAttach event
2021-09-07 13:06:19,997 OnAttached event for device: acA3088-16gc
2021-09-07 13:06:19,997 OnOpen event for device: acA3088-16gc
2021-09-07 13:06:20,031 OnOpened event for device: acA3088-16gc
2021-09-07 13:06:22,707 OnAttach event
2021-09-07 13:06:22,707 OnAttached event for device: acA3088-16gc
2021-09-07 13:06:22,707 OnOpen event for device: acA3088-16gc
2021-09-07 13:06:22,748 OnOpened event for device: acA3088-16gc
2021-09-07 13:06:25,431 OnAttach event
2021-09-07 13:06:25,431 OnAttached event for device: acA3088-16gc
2021-09-07 13:06:25,432 OnOpen event for device: acA3088-16gc
2021-09-07 13:06:25,473 OnOpened event for device: acA3088-16gc
2021-09-07 13:06:28,156 OnAttach event
2021-09-07 13:06:28,157 OnAttached event for device: acA3088-16gc
2021-09-07 13:06:28,157 OnOpen event for device: acA3088-16gc
2021-09-07 13:06:28,197 OnOpened event for device: acA3088-16gc
2021-09-07 13:06:30,709 OnGrabStart event for device: front_bc_bottom
2021-09-07 13:06:30,724 OnGrabStarted event for device: front_bc_bottom
2021-09-07 13:06:30,725 OnGrabStart event for device: front_bc_top
2021-09-07 13:06:30,740 OnGrabStarted event for device: front_bc_top
2021-09-07 13:06:30,740 OnGrabStart event for device: front_bc_middle
2021-09-07 13:06:30,755 OnGrabStarted event for device: front_bc_middle
2021-09-07 13:06:30,756 OnGrabStart event for device: top_close
2021-09-07 13:06:30,772 OnGrabStarted event for device: top_close
2021-09-07 13:06:30,773 OnGrabStart event for device: left_side
2021-09-07 13:06:30,788 OnGrabStarted event for device: left_side
2021-09-07 13:06:30,789 OnGrabStart event for device: right_side
2021-09-07 13:06:30,805 OnGrabStarted event for device: right_side
2021-09-07 13:06:30,805 OnGrabStart event for device: top_far
2021-09-07 13:06:30,821 OnGrabStarted event for device: top_far
2021-09-07 13:06:30,822 OnGrabStart event for device: front
2021-09-07 13:06:30,837 OnGrabStarted event for device: front
2021-09-07 22:29:25,911 Camera device left_side disconnected!
2021-09-08 02:11:12,491 Camera device front_bc_middle disconnected!
```
It is not always the same cameras that disconnects and the time between the system getting idle and the first OnCameraDeviceRemoved event fluctuates. 

Now my question is: Is there anything that I can do about this, either in the configuration of the cameras or in my code like e.g. reconnect the removed/disconnected devices? Unfortunately I did not find anything on this topic in the pypylon examples nor in the documentation...

Note: The cameras are still accessible via the network, as they are pinged by my program once a minute and this monitoring does not indicate an error.

Thanks in advance,

Jens

SMA2016a, 2021-09-11:
1) first of all try to disable power saving options in BIOS as well as in Linux OS.
2) check with Wireshark or tcpdump if the heartbeat timeout signal will be reset frequently. By default it is 1sec. Set it manually for 5 min.

This is written in pylon c++ programmers guide in advance topics.

jsiebert, 2021-09-14:
Hey @SMA2016a,

thanks for the hint! I'll test your suggestion this week. Closing this ticket for now...

closed 2021-09-14 by jsiebert

mehmetgencg, 2022-01-27:
Hello @jsiebert , 

I am also trying to make a system which grabs images using a hardware trigger and I want to get a log when a camera is removed. Could you share your source code with me?

jsiebert, 2022-01-27:
Hey @mehmetgencg,

unfortunately, I cannot share my source code as it is part of a closed source commercial application.

But my code is based on examples from the "[samples](https://github.com/basler/pypylon/tree/master/samples)" directory in the pypylon source repository, e.g. "[configurationeventprinter.py](https://github.com/basler/pypylon/blob/master/samples/configurationeventprinter.py)" and "[grabcameraevents.py](https://github.com/basler/pypylon/blob/master/samples/grabcameraevents.py)".

There you can find examples on how to write an event handler which is able to detect whether a camera is still connected to your application and how to register such an event handler with your application.

mehmetgencg, 2022-01-27:
@jsiebert  Thank you for your help :) 
My problem is I stuck with calling the events. How I should call it when I use cam_array for multiple cameras.

jsiebert, 2022-01-27:
In case with "calling the events" you mean that you would like to register the event handler for multiple cameras, you can iterate over your `cam_array` and call `cam.RegisterConfiguration(eventhandler, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)` for each camera device whose information is stored in the array.

The individual event handler methods in the event handler will be called by the pypylon API as soon as some event occurs in your system. You can get the information about which camera triggered the event using the `camera` parameter of the respective event handler method.