basler/pypylon #409: How to display multiple-Basler Camera on a GUI made through tkinter in python?

Jondila, 2021-09-10:
I am tying to create a gui in python that will display the images based on the number of cameras.
I have made the codes for 4 that runs properly and made a gui that runs properly. The problem is to link those for cameras to the GUI.
Here are my codes

```
import os
os.environ["PYLON_CAMEMU"] = "3"
from pypylon import genicam
from pypylon import pylon
import cv2
import sys
import numpy as np
import time
```

```
from pypylon import pylon
tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices()
for device in devices:
    print(device.GetFriendlyName())
```

```
countOfImagesToGrab = 100
maxCamerasToUse = 4
exitCode = 0
a = 0

try: 
    
    tlFactory = pylon.TlFactory.GetInstance()
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RUNTIME_EXCEPTION("No camera present.")

    cameras = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))
    l = cameras.GetSize()
    for i, cam in enumerate(cameras):
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        print("Using device ", cam.GetDeviceInfo().GetModelName())

    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
    cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

    while cameras.IsGrabbing():

        grabResult0 = cameras[0].RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        grabResult1 = cameras[1].RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        grabResult2 = cameras[2].RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        grabResult3 = cameras[3].RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

   
            
        if grabResult0.GrabSucceeded() and grabResult1.GrabSucceeded() and grabResult2.GrabSucceeded() and grabResult3.GrabSucceeded():
            image0 = converter.Convert(grabResult0)
            img0 = image0.GetArray()
            image1 = converter.Convert(grabResult1)
            img1 = image1.GetArray()
            image2 = converter.Convert(grabResult2)
            img2 = image2.GetArray()
            image3 = converter.Convert(grabResult3)
            img3 = image3.GetArray()
            
            
            img0_s = cv2.resize(img0, dsize);img1_s = cv2.resize(img1, dsize)
            #img2_s = cv2.resize(img2, dsize);img3_s = cv2.resize(img3, dsize)
            

            
            # show 4 imgs
            cv2.imshow('camera3', img0_s)
            cv2.imshow('camera1', img1_s)
            cv2.imshow('camera4', img2_s)
            cv2.imshow('camera2', img3_s)

            # LED USB -turn on : save img / else "NULL"
            k = cv2.waitKey(1)
            
            start_1 = time.time()            
         
            if(img1_s[hor1,ver1,0] > br_th):
                print("NULL")
                time.sleep(wt)
            else:

                # save 4 imgs
                #cv2.imwrite('./camera3%a.jpg'%a,img0_s) #img0
                #cv2.imwrite('./camera1%a.jpg'%a,img1_s) #img1
                #cv2.imwrite('./camera4%a.jpg'%a,img2_s) #img2
                #cv2.imwrite('./camera2%a.jpg'%a,img3_s) #img3
                
                # for check
                print("Save")
                print(a)                
                end_1= time.time()
                #print('time elapsed:', end_1 - start_1)                
                a = a+1
                time.sleep(wt)

            
            # ESC키 누르면 종료
            if k == 27:
                cv2.destroyAllWindows()
                print("ECS")
                break
                
            #time.sleep(1)

        grabResult0.Release()
        grabResult1.Release()
        #grabResult2.Release()
        #grabResult3.Release()

    #cameras.StartGrabbing()
    cv2.destroyAllWindows()            
except genicam.GenericException as e:
    # Error handling
    #print("An exception occurred.", e.GetDescription())
    exitCode = 1

sys.exit(exitCode)
```

```
from tkinter import *
from PIL import Image,ImageTk
from tkinter import filedialog
from PIL import Image, ImageTk
import os
import cv2
import numpy as np
import time

class application(Tk):
    def __init__(self,parent):
        Tk.__init__(self,parent)
        self.parent = parent
        self.minsize(width=1000,height=800)
        self.initialize()
        
    def initialize(self):
        self.grid_columnconfigure(0,weight=1)
        self.grid_columnconfigure(1,weight=1)
        self.grid_columnconfigure(2,weight=1)
        
        #Text widgets
        ###########################################################################################
        self.text1 = Text(self,width=30,height=2, state=NORMAL)
        self.text1.grid(column=0,row=1,sticky='W',padx=680)
        
        self.text2 = Text(self,width=30,height=2, state=NORMAL)
        self.text2.grid(column=0,row=2,sticky='W',padx=680)
        
        self.text3 = Text(self,width=30,height=2, state=NORMAL)
        self.text3.grid(column=0,row=3,sticky='W',padx=680)
        
        self.text4 = Text(self,width=30,height=2, state=NORMAL)
        self.text4.grid(column=0,row=4,sticky='W',padx=680)
        
        self.text5 = Text(self,width=30,height=2, state=NORMAL)
        self.text5.grid(column=0,row=5,sticky='W',padx=680)
        
        #Canvas widget for inserting result
        ############################################################################################
        self.canvas1 = Canvas(self,height=200,width=200,state=NORMAL)
        self.canvas1.grid(column=0,row=2,sticky='W',padx=50)
        
        self.canvas2 = Canvas(self,height=200,width=200,state=NORMAL)
        self.canvas2.grid(column=0,row=2,sticky='W',padx=300)
        
        self.canvas3 = Canvas(self,height=200,width=200,state=NORMAL)
        self.canvas3.grid(column=0,row=4,sticky='W',padx=50)
        
        self.canvas4 = Canvas(self,height=200,width=200,state=NORMAL)
        self.canvas4.grid(column=0,row=4,sticky='W',padx=300)
        
        
        #Label widget
        ############################################################################################
        self.label1 = Label(self,text="Images",width=10,height=2,anchor="w")
        self.label1.grid(column=0,row=0,sticky='W',padx=250)

        
        self.label2 = Label(self,text="Camera1",bg='green',width=10,height=2,anchor="w")
        self.label2.grid(column=0,row=1,sticky='W',padx=100)
        
        self.label3 = Label(self,text="Camera2",bg='green',width=10,height=2,anchor="w")
        self.label3.grid(column=0,row=1,sticky='W',padx=350)
        
        self.label4 = Label(self,text="Camera3",bg='green',width=10,height=2,anchor="w")
        self.label4.grid(column=0,row=3,sticky='W',padx=100)
        
        self.label5 = Label(self,text="Camera4",bg='green',width=10,height=2,anchor="w")
        self.label5.grid(column=0,row=3,sticky='W',padx=350)
        
        #Button widget
        ############################################################################################
        
        self.button21= Button(self,text='Exit',bg='green',width=10,height=2,command=self.destroy)
        self.button1.grid(column=0,row=6,sticky='S', pady=20)
        
if __name__ == "__main__":
    app = application(None)
    #font.nametofont('TkDefaultFont').configure(size=10)
    app.title("Four Camera")
    app.configure(bg='gray')
    app.geometry('1000x800')
    app.mainloop()
```
I tried to follow some threads including [this](https://github.com/basler/pypylon/issues/72). I couldn't get through either.

SMA2016a, 2021-09-11:
have you tried this?
these examples seem to be very promising :

https://pythonguides.com/python-tkinter-image/

Jondila, 2021-10-21:
I have solved the problem by using threading procedures,  and initiating global variables. this can be easily accessed through accessing the following issue https://github.com/basler/pypylon/issues/425