basler/pypylon #410: pip install failing

jpiabrantes, 2021-09-15:
Pip install is failing on python 3.9.6. MacPro M1 Chip. Any tips?

```sh
> pip install pypylon
ERROR: Could not find a version that satisfies the requirement pypylon (from versions: none)
ERROR: No matching distribution found for pypylon
```

SMA2016a, 2021-09-19:
as written in the requirement list, you need to have the latest pip version.
update your pip version. 
python -m pip install --upgrade pip

thiesmoeller, 2021-09-19:
Unfortunately we haven't released pylon SDK for Apple silicon.
Meanwhile you have to run in rosetta2

See https://github.com/basler/pypylon/issues/322#issuecomment-783310781

thiesmoeller, 2023-06-14:
Release 2.2.0 with full support of macOS x86_64 and arm64 ( >= macOS 11 ) is out.

Available via pip install or releases page.

closed 2023-06-14 by thiesmoeller