basler/pypylon #411: CameraFinder & Python doc

maximecharriere, 2021-09-21:
In C# there is a CameraFinder to enumarate all connected cameras and emulators.
Is there an equivalence in Python ?

--- 

As explained in some places, I can use the C++ doc for the Python wrapper. But I don't understand because many C++ examples don't work.
### For example for camera emulation with custom images :
**C++ doc**
```c++
// ** Custom Test Images **
// Disable standard test images
camera.TestImageSelector.SetValue(TestImageSelector_Off);
// Enable custom test images
camera.ImageFileMode.SetValue(ImageFileMode_On);
// Load custom test image from disk
camera.ImageFilename.SetValue("c:\\images\\image1.png");
```
**What I coded in Python based on the C++ doc and did not work**
```python
// ** Custom Test Images **
// Disable standard test images
camera.TestImageSelector.SetValue(pylon.TestImageSelector_Off)
// Enable custom test images
camera.ImageFileMode.SetValue(pylon.ImageFileMode_On)
// Load custom test image from disk
camera.ImageFilename.SetValue("c:\\images\\image1.png")
```
**What I had to code in Python to make it work**
```python
// ** Custom Test Images **
// Disable standard test images
camera.TestImageSelector = "Off"
// Enable custom test images
camera.ImageFileMode = "On"
// Load custom test image from disk
camera.ImageFilename = "c:\\images\\image1.png"
```

Where can I find what to write to call methods and get/set values ? How do I know what methods and enums are available in python and how to write them ?

SMA2016a, 2021-09-22:
1)  Try following code 
Please refer to following code
https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py

SMA2016a, 2021-09-22:
2) Some parameter is not available on across the camera models. So you need to check with pylonviewer if the parameter available on your camera.

or via SDK e.g 
if genicam.IsAvailable(camera.Exposuretime):

thiesmoeller, 2021-09-22:
See also https://github.com/basler/pypylon-samples
for in depth samples and a webinar