basler/pypylon #415: how to save images captured from 3 cameras at the same time into 3 drives using the same code????????????????

thasnimolvs, 2021-10-01:
the images from 3 cameras saving to 3 different  drives ,

SMA2016a, 2021-10-10:
@thasnimolvs 
please describe your question in detail or if you already solved the issue , please close it then.

as I understood your question, this is just a matter of managing the filename 

paths= ["C:\images", "D:\Images", "E:\images"]

Now you can call RetrieveResult on all cameras and save in corresponding directory.

thiesmoeller, 2025-12-15:
This question about saving images from multiple cameras to different drives has been addressed in the comments with code examples. We are closing this issue. For further questions on multiple cameras, please refer to the discussion or open a new issue.