basler/pypylon #416: Obtaining current FPS

jokubas11, 2021-10-04:
As I saw in previous threads, there is no documentation for pypylon and people are being referred to Basler page that contains documentation for C++ only. Anyway, most of the way I was getting along well, until I needed to get the framerate from my camera.

The camera does support resulting framerate command. 
I have tried following:
`print(myCameraObject.ResultingFrameRate.GetValue())`
I have also tried:
`print(myCameraObject.ResultingFrameRate`
also
`fps = myCameraObject.ResultingFrameRate`
`print(fps)`
or things like that... None work... Can someone provide me with correct command, as there is no documentation for pypylon and I am out of ideas.

The error I am getting:
`Node not existing (file 'genicam_wrap.cpp', line 16130)`

thiesmoeller, 2021-10-04:
Please  check  our model specific documentation at docs.baslerweb.com

The camera features do come from the camera itself. Pylon/PyPylon is just presenting them as API.

For background see the webinar and more samples at https://github.com/basler/pypylon-samples.

jokubas11, 2021-10-05:
I have said that I did check the documentation and did find the corresponding command, however it does not work, therefore I am looking for the command that I need to call in Python to invoke that feature. I have found my command in Basler docs. The documents are not helpful, however.

thiesmoeller, 2021-10-05:
@jokubas11 what is your camera model?

closed 2021-10-12 by thiesmoeller