basler/pypylon #421: Failed to receive frame from basler camera for several consecutive frames

MiladGhorbaniG, 2021-10-15:
Hi @rgov 
I am using a Jetson Nano with OpenCV 4.5 to read the Basler camera every 50 ms. I call the following function: 

`def Cam_one(w,h):
    # global camera_success,image_success
    global camera_success,image_success,camera,converter
    camera_success=0
    image_success=0
    if camera.IsGrabbing():
        camera_success=1
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():
            image_success=1
            # Access the image data
            image = converter.Convert(grabResult)
            im = image.GetArray()
            img=cv2.resize(im,(w,h))
            # sttt = time.time()
            img_encoded = cv2.imencode('.jpg', img)[1].tobytes()  # bytes class

            # 'repair' image from byte array
            nparr = np.frombuffer(img_encoded, np.byte)
            img = cv2.imdecode(nparr, cv2.IMREAD_ANYCOLOR)
            # print('delay conversion jpeg = ' , time.time()-sttt)
            print(img.shape)
        else:
            image_success=0
        grabResult.Release()
    else:
        camera_success=0
    return image_success , img`

About 1 percent of the time, it fails for several consecutive frames, and I can receive a new frame for example 700 ms later.

SMA2016a, 2021-10-31:
1) first, measure the time taken for the following block.

grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grabResult.GrabSucceeded():
            image_success=1
            # Access the image data
            image = converter.Convert(grabResult)
            im = image.GetArray()
            img=cv2.resize(im,(w,h))
        else:
            image_success=0
        grabResult.Release()
    else:
        camera_success=0

2) Then set the AOI to 640*480  and check if the issue is still occurring.
3) since you are already using opencv, try also use opencv for image conversation too.

thiesmoeller, 2025-12-15:
This issue about occasional frame failures requires more information about error handling, buffer settings, and system performance. Since we haven't received updates, we are closing this issue. If you are still experiencing frame failures, please open a new issue with detailed error logs and system information.