basler/pypylon #432: EnumerateDevices does not grab all connected cameras

kimdonggyun, 2021-11-02:
Simple error.
I connected two cameras and pylon IP confabulator shows two cameras connected.

And to check the cameras through pypylon,
```
tl_factory = pylon.TlFactory.GetInstance()
set_devices = tl_factory.EnumerateDevices()
print(len(set_devices))
```
result is 1 (one camera connected)

I've checked Serial Number of connected camera through pypylon and the SN appeared randomly.

SMA2016a, 2021-11-03:
-What is the OS?
-Did you set the static IP address to the camera?
-are the cameras connected to a network switch?

kimdonggyun, 2021-11-03:
> -What is the OS?
> -Did you set the static IP address to the camera?
> -are the cameras connected to a network switch?

I'm using Mac OS Monterey
I tried both Auto and DHS IP address
Two cameras are connected to a network hub and a ethernet cable connects between Mac and the network hub (where both cameras are connected as well)

closed 2021-11-10 by kimdonggyun

SabraHashemi, 2022-05-15:
@kimdonggyun  did you solve this?

SMA2016a, 2022-05-15:
@SabraHashemi do you also use multiple camera? and same issue?   if yes, would send me a tcpdump via private message, please?

kimdonggyun, 2022-05-16:
Hello, @SabraHashemi Yes I solve this problem. But it is not an official way. I am using two different camera models and with the model having a USB connection, both cameras were detected immediately. However, ethernet cameras need some scripts to check whether two cameras are connected or not (it always takes some time to connect two cameras and sometimes you even have to reconnect the cameras physically). I will share the code if you like.

Alextraccar, 2022-08-15:
@kimdonggyun Hi Kim, I am experiencing the same problem of enumerating 2 GigE cams.  Could you share the code you commented above that some scripts are needed to check whether two cameras are connected or not.   Thanks.

Alextraccar, 2022-08-15:
@SabraHashemi Did you solve the issue by just setting 2 GigE cams to DHCP or did you also make any other changes?  Did you also set the computer network adapter to DHCP?  Thanks.

kimdonggyun, 2022-08-16:
Hello @SabraHashemi 
This way, I checked the two cameras connected. 

```
from pypylon import pylon

def cam_init (dev_number):
    """
    getting instance of basler cam
    return connected camera instances
    """
    print("getting the first camera instance")
    tl_factory = pylon.TlFactory.GetInstance() # get the first instance with all connected cameras
    devices = tl_factory.EnumerateDevices()

    dev_info(devices) # check whether cameras are connected

    set_tl_factory, set_devices = cam_non_identical(tl_factory, devices, dev_number) # check all physically conneceted cameras get instance

    cams = dev_2_array (set_tl_factory, set_devices, dev_number)

    return cams

def cam_non_identical (tl_factory, devices, dev_number):
    """
    Basler's package have a problem with MacOS and sometimes all connected camera doesn't show up.
    This def is created to check whether call camera's instants are created.
    Therefore, no duplicating camera instant.
    This function will be running untill you get certain number of camera instant you desire
    devices = initial camera instant
    dev_number = number of devices connected
    """
    
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera connected, please check the camera connection!")

    else:
        temp_dev_list = [] # create empty camera list to com
        for dev_info in devices:
            temp_dev_list.append(dev_info.GetSerialNumber())
        
        if len(set(temp_dev_list)) == dev_number: # check unique number of device in the devices list
            print("all cameras' instance are created", temp_dev_list)
            return tl_factory, devices

        else:
            while len(set(temp_dev_list)) != dev_number: # count the unique camera instant and compare it with desired number of cameras
                tl_factory = pylon.TlFactory.GetInstance() # get instance again
                devices = tl_factory.EnumerateDevices()
                temp_dev_list = []
                for dev_info in devices:
                    temp_dev_list.append(dev_info.GetSerialNumber())
                print(temp_dev_list)
                if len(set(temp_dev_list)) == dev_number:
                    print("all cameras' instance are created", temp_dev_list)
                    break

            return tl_factory, devices

def dev_info(devices):
    """
    print out all connected cameras' basic information
    """
    if len(devices) == 0:
        raise pylon.RuntimeException("No camera connected")
    else:
        print("total number of devices:", len(devices)) # print total number of devices connected
        for i, dev_info in enumerate(devices):
            # for GigE cameras: dev_info.GetDeviceClass() == "BaslerGigE"
            # for USB cameras: dev_info.GetDeviceClass() == "BaslerUsb"
            print("using %s as IP %s SN %s" % (dev_info.GetModelName(), dev_info.GetIpAddress() ,dev_info.GetSerialNumber()))
```

Alextraccar, 2022-08-16:
@kimdonggyun Thank you very much for the help.  The definition of function dev_2_array is missing, could you also provide that?  Thank you again.

Alextraccar, 2022-08-16:
@kimdonggyun Hi Kim, I have tested your code and what I found is that as the function of EnumerateDevices() is buggy (which is why we both had the issue in the first place where EnumeratsDevices() does not show all connected GigE cams), it always only show just 1 of the 2 connected GigE cams, your code actually stuck in the following while loop inside the cam_non_identical function and keeps printing out the SN of the randomly enumerated 1 of the 2 cams.   Have you tried your code with 2 GigE cams and successfully enumerated both cams?  Thanks.

while len(set(temp_dev_list)) != dev_number: 
                tl_factory = pylon.TlFactory.GetInstance() # get instance again
                devices = tl_factory.EnumerateDevices()
                temp_dev_list = []
                for dev_info in devices:
                    temp_dev_list.append(dev_info.GetSerialNumber())
                print(temp_dev_list)
                if len(set(temp_dev_list)) == dev_number:
                    print("all cameras' instance are created", temp_dev_list)
                    break

thiesmoeller, 2022-08-16:
please use the proper multi line code markdown. This makes your comments way more readable:

````
```python
<Your code>
```
````

Can you please check https://github.com/basler/pypylon/issues/159#issuecomment-836537265 
This bug hasn't been fixed until now on macos. So you might run into the same issue

Alextraccar, 2022-08-16:
@thiesmoeller thanks for your comments.  I don't think we are referring to the same bug.  I guess you are refereeing to the bug that EnumerateDevices() always show duplicate camera instances while the bug I am experiencing is same as what @kimdonggyun experienced that when two GigE cams were connected, EnumerateDevices() only randomly show 1 of the 2 cams connected.