basler/pypylon #433: multi camera video capturing by using python's multiprocessing

kimdonggyun, 2021-11-02:
As my camera module is not applicable for the "Action Commands (capturing images at the same time from multiple cameras)", I was looking for another method to do simultaneous video recording from multiple cameras (in my case only two cameras).

I wrote a definition to record and save the video like below:
```
#function for simple video recording
def video_recording(file_dir, filename, caminstance):
    with get_writer(os.path.join(file_dir, filename)) as writer:
        print("recording start with %s at %s" % (cam.DeviceInfo.GetFriendlyName(), datetime.now()))
        caminstance.StopGrabbing()
        caminstance.StartGrabbingMax(100)
        while caminstance.IsGrabbing():
            res = caminstance.RetrieveResult(100)
            writer.append_data(res.Array)
            res.Release()
        print("recording finish with %s at %s" % (cam.DeviceInfo.GetFriendlyName(), datetime.now()))
```

And by using python's multiprocessing method, I tried to run the "video_recording" function at the same time.

```
cam1 = Process(target=video_recording, 
                args=("file_path_dir", "test_multicore_%s.avi" % (video_pos[0]), cams[0]))
cam2 = Process(target=video_recording, 
                args=("file_path_dir", "test_multicore_%s.avi" % (video_pos[1]), cams[1]))
cam1.start()
cam2.start()
cam1.join()
cam2.join()
```

when I ran the function individually, it took around 30 sec for each camera (I set like that.), and saved the video successfully. Therefore, the function itself is working.

However, when I ran it with multiprocessing, the function ended early and stopped at:
```
        while caminstance.IsGrabbing():
            res = caminstance.RetrieveResult(100) <----- here
            writer.append_data(res.Array)
            res.Release()
```

Therefore no video recording was done.

Is there anyway I can use multiprocessing for the simultaneous video capturing?

kimdonggyun, 2021-11-04:
I added full script so maybe someone can help practically.

```
# list cameras connected with PC
# 0. set number of cameras to use
cameras_2_use = 2

# 1. get transport layer and all attached devices
tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices() #enumerate devices connected

# 1.1 soft ware issue from pylon. this block may be deleted later if the issue is solved
# here, get instance and enumerate devices until two different cameras are recognized
temp_dev_list = []
if len(devices) == 0:
    raise pylon.RuntimeException("No camera connected")
else:
    for i, dev_info in enumerate(devices):
        temp_dev_list.append(dev_info.GetSerialNumber())
    while temp_dev_list[0] == temp_dev_list[1]:
        tl_factory = pylon.TlFactory.GetInstance()
        devices = tl_factory.EnumerateDevices()
        temp_dev_list = []
        for i, dev_info in enumerate(devices):
            temp_dev_list.append(dev_info.GetSerialNumber())
        print(temp_dev_list)
        if temp_dev_list[0] != temp_dev_list[1]:
            break

print(temp_dev_list)
"""
# 1.1 software issue from pylon. this block may be deleted later
# remove duplicated camera list
set_devices = []
for device in devices:
    if device not in set_devices:
        set_devices.append(device)
    else:
        pass
set_devices = tuple(set_devices)
"""

# 2. check the devices information
if len(devices) == 0:
    raise pylon.RuntimeException("No camera connected")
else:
    print("total number of devices:", len(devices)) # print total number of devices connected
    for i, dev_info in enumerate(devices): # 
        if dev_info.GetDeviceClass() == 'BaslerGigE': # check if the connected cameras are GigE model. Otherwise software trigger won't work
            print("using %s as IP %s SN %s" % (dev_info.GetModelName(), dev_info.GetIpAddress() ,dev_info.GetSerialNumber()))
        else:
            raise EnvironmentError("no GigE device found")

# 3. create instant array and attach all devices on the Array
cams = pylon.InstantCameraArray(min(len(devices), cameras_2_use)) # create instant
for i, cam in enumerate(cams):
    cam.Attach(tl_factory.CreateDevice(devices[i]))

# setting camera's parameters
fps = 20 # frame per sec
ExposureTime = 2000 # in µs
Height, width = 962, 1286 # in pixel

for cam in cams:
    cam.Open() # open camera to change the parameter
    print("Setting device ", cam.GetDeviceInfo().GetFriendlyName())
    #print(cam.MaxNumBuffer.GetValue(), cam.GetQueuedBufferCount(), cam.MaxNumQueuedBuffer.GetValue())
    cam.Height.SetValue(Height)
    cam.Width.SetValue(width)
    cam.ExposureTimeRaw.SetValue(ExposureTime)
    cam.AcquisitionFrameRateAbs.SetValue(fps)
    print(cam.Height.GetValue(), cam.Width.GetValue(), cam.ExposureTimeRaw.GetValue(), cam.AcquisitionFrameRateAbs.GetValue())

    cam.Close() # close camera

# capturing video with multiple cameras at simultaneously
video_pos = {0:"front", 1:"side"}

#function for simple video recording
def video_recording(file_dir, filename, caminstance):
    with get_writer(os.path.join(file_dir, filename)) as writer:
        print("recording start with %s at %s" % (caminstance.DeviceInfo.GetFriendlyName(), datetime.now()))
        caminstance.StopGrabbing()
        caminstance.StartGrabbingMax(50)
        while caminstance.IsGrabbing():
            try :
                print("working")
                res = caminstance.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
                print(res)
            except:
                print("something wrong while recording")
            writer.append_data(res.Array)
            res.Release()
        print("recording finish with %s at %s" % (caminstance.DeviceInfo.GetFriendlyName(), datetime.now()))

cam1 = Process(name="cam1", target=video_recording, 
                args=("/Users/dkim/Desktop/basler_camera/recording", "test_multicore_%s.avi" % (video_pos[0]), cams[0]))
cam2 = Process(name="cam2", target=video_recording, 
                args=("/Users/dkim/Desktop/basler_camera/recording", "test_multicore_%s.avi" % (video_pos[1]), cams[1]))
cam1.start()
cam2.start()
```

closed 2021-11-10 by kimdonggyun

alexjpm, 2022-04-19:
Hi @kimdonggyun. Were you able to solve this? Think I'm bumping into something similar.

kimdonggyun, 2022-04-21:
Hi, @alexjpm. Yes, I could record the video from two cameras simultaneously. Technically, it's not really simultaneous, but the timestamp between two cameras is really neglectable. What is your problem? Maybe you can describe your issue shortly.

fxyQAQ, 2022-08-31:
Hi, @kimdonggyun. I want to realize the data acquisition display and saving of 8 webcams, but I don't know much about the mechanism of multiprocessing, would you mind sharing the code with me?

kimdonggyun, 2022-09-01:
> Hi, @kimdonggyun. I want to realize the data acquisition display and saving of 8 webcams, but I don't know much about the mechanism of multiprocessing, would you mind sharing the code with me?

Hi @fxyQAQ . The code is above. The code is for only two cameras. But you can easily replace certain parts to optimize for 8 cameras.