basler/pypylon #434: pip install failing on Windows 11

rddunphy, 2021-11-05:
I'm having trouble installing pypylon on a Windows 11 machine with Python 10 and pip version 21.3.1.

```
> py -m pip install pypylon
ERROR: Could not find a version that satisfies the requirement pypylon (from versions: none)
ERROR: No matching distribution found for pypylon
```

I see from a number of similar issues that this is usually caused by not having an up-to-date version of pip, but in this case I believe I have the latest version, and running `py -m pip install --upgrade pip` results in "requirement already satisfied". Are there any other known issues that could be causing this?

SMA2016a, 2021-11-14:
I think that pypylon current version isn't released  and tested on Windows 11. 
do you install all stuff directly on Windows environment or use any virtual environment?

Do you mean python 3.10?

as you can see in the table, the latest tested python version is 3.9, may be that is the issue.
https://github.com/basler/pypylon

rddunphy, 2021-11-21:
I did indeed mean Python 3.10, and this was the cause of the problem - after downgrading to Python 3.9 it works fine. For some reason, I was getting warnings about the Pip version even after upgrading it, and I was so focused on figuring out the Pip requirement that I completely missed the Python requirements - sorry about that!

closed 2021-11-21 by rddunphy

xiaoxstz, 2022-01-20:
I also want to use python 3.10. When will it support Python3.10?