basler/pypylon #435: Make camera always use same UDP port

jokubas11, 2021-11-05:
I am using camera acA2040-25gmNIR and it uses UDP ports for data to travel. Issue is that windows firewall blocks UDP traffic by default. Turning off firewall sorts the issue, but that is not suitable for me as a permanent solution. I was trying to write a firewall rule that would permit the traffic from UDP port. I have permitted all UDP ports and it solves an issue, however this is not suitable as a permanent solution. I would like to write a firewall rule that permits data transfer only from one specific UDP port which would be the camera in question. However, every time I run an app, the camera uses different UDP port. Is there a way to make camera always use specific UDP port so I can write firewall exception for this port?

thiesmoeller, 2021-11-05:
See #429

thiesmoeller, 2021-11-22:
@jokubas11: is it working for you this way?