basler/pypylon #436: cannot pickle 'SwigPyObject' object

kimdonggyun, 2021-11-10:
While I'm running multiprocessing to handle multiple cameras, I got following error message
"cannot pickle 'SwigPyObject' object'

Does it mean pypylon's image processing is not pickable?
Otherwise, is there anyway I can solve this issue?

```
# capturing video with multiple cameras simultaneously
video_pos = {0:"front", 1:"side"}

#function for simple video recording
def video_recording(file_dir, filename, caminstance):
    #caminstance.StopGrabbing()
    caminstance.StartGrabbingMax(200)
    with get_writer(os.path.join(file_dir, filename)) as writer:
        print("parent process : %s / process id %s" % (os.getppid(), os.getpid()))
        print("recording start with %s at %s" % (caminstance.DeviceInfo.GetFriendlyName(), datetime.now()))
        caminstance.StopGrabbing()
        caminstance.StartGrabbingMax(50)
        while caminstance.IsGrabbing():
            try :
                print("working")
                res = caminstance.RetrieveResult(10000, pylon.TimeoutHandling_ThrowException)
                print(res)
            except:
                print("something wrong while recording")
            writer.append_data(res.Array)
            res.Release()
        print("recording finish with %s at %s" % (caminstance.DeviceInfo.GetFriendlyName(), datetime.now()))
        caminstance.Close()

cam1 = Process(name="cam1", target=video_recording, 
                args=("/Users/dkim/Desktop/basler_camera/recording", "test_multicore_%s.avi" % (video_pos[0]), cams[0]))
cam2 = Process(name="cam2", target=video_recording, 
                args=("/Users/dkim/Desktop/basler_camera/recording", "test_multicore_%s.avi" % (video_pos[1]), cams[1]))

cam1.start() <----- I got error message at this line
cam2.start()

cam1.join()
cam2.join()
```

thiesmoeller, 2021-11-10:
Hi @kimdonggyun, 

Multiprocessing with binary extensions is typically tricky in python and with pypylon not supported except special cases.

What do you want to realize by using multiprocessing?

As all capturing inside pypylon is already multthreaded  ( real OS threads ) the image acquisition from multiple cameras is not limited by python language...

It is possible to run high performance multi camera applications without multiprocessing

kimdonggyun, 2021-11-10:
What I want to get from multiprocessing is two videos from two cameras. Important thing is two camera should be starting and ending exactly same time.

I would like to use the other multi camera application but so far python is the only tool that I can use.

Anyway, I tried threading instead of multiprocessing. And now it records the video although there are some issues such as horizontal noise and camera doesn't capture the image e.g. the array is just 0 shape.

I will open another issue for these issues.

closed 2021-11-10 by kimdonggyun

Alextraccar, 2022-12-09:
Hi Kim, I am also trying to apply multiprocessing to 3 GigE cams for concurrent image collection and also experienced with the same "cannot pickle SwigPyObject" error.  You mentioned that you solved the problem by switching to threading. Can you share the code of applying threading to achieve the parallel/synchronization operation of 2 cameras?  Thank you in advance.

thiesmoeller, 2023-02-15:
Don't transfer the full camera object to multiple processes but transfer the image data to the workers instead.
Collect in main process work in child processes....
See this example here https://github.com/basler/pypylon/issues/513#issuecomment-1346405311