basler/pypylon #437: horizontal band noise while recording

kimdonggyun, 2021-11-10:
While video was recording through pylon viewer, sometimes there are horizontal band noise was generated on the video, quite randomly.
How can I avoid this horizontal band noise?
Is it due to my computational power (e.g. buffer size)?

SMA2016a, 2021-11-11:
do you USB or GigE cameras?

GigE: set the parameter interpacket delay
USB: reduce the value of DeviceLinkThroughputLimit .

If this does not help you. Send a screen shoot that shows the issue.

kimdonggyun, 2021-11-11:
>  interpacket delay

I am using GigE cameras.
What would be the possible effect on the camera if I set the inter packet delay parameter?
And what would be the optimal or default value of inter packet delay?

kimdonggyun, 2021-11-11:
> do you USB or GigE cameras?
> 
> GigE: set the parameter interpacket delay USB: reduce the value of DeviceLinkThroughputLimit .
> 
> If this does not help you. Send a screen shoot that shows the issue.

I set the lnter package delay parameter and it solved all problems (dropped image, horizontal band noise)

Many thanks

closed 2021-11-11 by kimdonggyun