basler/pypylon #440: How to detect when a USB camera gets plugged in

robotrapta, 2021-12-01:
I'm trying to build a system that detects when a Basler ACE camera gets plugged in, and starts using it.  My code does this:

```
tlf = pylon.TlFactory.GetInstance()
devices = tlf.EnumerateDevices()
if not devices:
    raise RuntimeException("No camera present")
```

If no cameras are present the first time I call this, then it seems to never notice if a camera gets plugged in later.  I've tried reloading the pylon library.  The shocking thing to me is that even if I exit the python process, and then restart it from a bash script, the new python process still will not detect the camera that has been plugged in.  The only way I can get pylon to detect a USB camera that gets plugged in after this process starts is to launch a new container, which is really problematic.

Is there some other way to detect if a camera is pressent in a way that doesn't "cache" the negative answer, or whatever is happening here?

thiesmoeller, 2021-12-01:
Can you post more details about your environment to find out what is not working in your docker setup:
Host OS and platform.
Simplified dockerfile + runtime settings that show your error.

On native host your intended operation is definitely working. Just open pylon viewer and plug in your cameras.

robotrapta, 2021-12-02:
The host is an ARM64 machine running Ubuntu 18 - NVIDIA's "tegra" flavor.  

```
$ uname -a
Linux smellycat 4.9.253-tegra #1 SMP PREEMPT Mon Jul 26 12:19:28 PDT 2021 aarch64 aarch64 aarch64 GNU/Linux
```

Using Python 3.8.10 and pypylon==1.7.2.dev6+pylon6.2.0 inside an Ubuntu 20.04 container running in microk8s.

I can work on a minimal repro.  I'm just flummoxed that restarting the process doesn't cause it to detect the device, but restarting the container does.  What could pypylon be leaving around after the process exits?  File handles?  Is there any part of the driver that should stay loaded in memory after the calling python process dies?

thiesmoeller, 2021-12-02:
Is your microk8s configured to start the container in privileged mode ?

Otherwise you won't get the udev events to detect the hot plugging of the camera from inside the container.

robotrapta, 2021-12-03:
Yes, it's in privileged mode.  For security, that's fine, and much easier than trying to manually map in the USB devices.

robotrapta, 2021-12-03:
Do I need to update the udev rules on the host?  Would that make a difference?  If so, why?  I hadn't because OS-defaults already include entries for the Basler ACE model we're using, so it didn't seem like it would change anything.

thiesmoeller, 2021-12-06:
if you `sudo udevadm monitor` in your container?
and un-plug and re-plug your camera, do you see the udev events? ( this would be the base for pypylon to detect your camera)

thiesmoeller, 2021-12-06:
This is a container configuration issue
Do you also mount the /dev/bus/usb directory, as recommended in typical tutorials on how to integrate usb devices?
In docker this would be a:
`docker run --privileged -v /dev/bus/usb/:/dev/bus/usb <image>`

thiesmoeller, 2021-12-10:
@robotrapta  -> does it work this way in _your_ microk8s environment?