basler/pypylon #442: How to avoid double frame grabbing in every initial  few frames of a 2-Basler Camera system using threading technique in Python?

Jondila, 2021-12-08:
My plan is implement a 4-10 Basler camera system using threading technique such that one camera facing the LED sensor could control any event of the rest of the cameras.
Suppose camera 1 faces the LED, if the selected point of interest in the live thread `camera1[y,x,0] >threshold`. Then camera 1 is not allowed to grab and save image, this situation will influence camera 2 also (will not grab and save image) at the same time.
I have made a GUI that combines the two cameras such that the whole implementation works without involving manual works. I have a problem with my `LiveThread` function that causes problems in my algorithm and I don't know how to solve it.

When I initialize camera and start live threading I found that camera 1 grabs two images at one likewise on camera 2. What I want is just one image per grabbing and not two images at per in every camera. With this problem the algorithm is violated and the program becomes useless. The following are the codes:

```
def InitCam():
        global maxCamerasToUse
        global camera
        global converter
        global bLiveThraed
        global panel
        global bCamInit
        global iterator

        try:
            # Get the transport layer factory.
            tlFactory = pylon.TlFactory.GetInstance()

            # Get all attached devices and exit application if no device is found.
            devices = tlFactory.EnumerateDevices()
            if len(devices) == 0:
                raise pylon.RUNTIME_EXCEPTION("No camera present.")

            # Create and attach all Pylon Devices.
            for i in range(len(devices)):
                if ((bCamOpen[0] == False)&(devices[i].GetSerialNumber() == '24011424')):
                    camera[0] = pylon.InstantCamera(tlFactory.CreateDevice(devices[i]))
                    print("Using device ", "CAM0:", camera[0].GetDeviceInfo().GetModelName(), camera[0].GetDeviceInfo().GetSerialNumber())

                elif ((bCamOpen[1] == False)&(devices[i].GetSerialNumber() == '24011514')):
                    camera[1] = pylon.InstantCamera(tlFactory.CreateDevice(devices[i]))
                    print("Using device ", "CAM1:", camera[1].GetDeviceInfo().GetModelName(), camera[1].GetDeviceInfo().GetSerialNumber())

            for i in range(maxCamerasToUse):
                # Open Camera
                if(bCamOpen[i] == False): 
                    camera[i].Open()
                    if(camera[i].IsOpen()): 
                        bCamOpen[i] = True
                        print("Cam",i,": Open")

                # Get Pixel format
                PixelFormat[i] = camera[i].PixelFormat.GetValue()

                # Set converter.OutputPixelFormat
                converter[i] = pylon.ImageFormatConverter()
                if(converter[i] == 'Mono8'):
                    converter[i].OutputPixelFormat = pylon.PixelType_Mono8
                else:
                    converter[i].OutputPixelFormat = pylon.PixelType_BGR8packed
                converter[i].OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        except genicam.GenericException as e:
            # Error handling
            print("An exception occurred.", e.GetDescription())
            exitCode = 1

def LiveThread(strIdx):
        CamIdx = int(strIdx)
        try:
            panel[CamIdx] = None
            image[CamIdx] = []
            # Start Grabbing
            camera[CamIdx].StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
            print("Cam",CamIdx,': Start Grabbing')

            while bLiveThraed[CamIdx]:
                grabResult = camera[CamIdx].RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

                if grabResult.GrabSucceeded():
                    time.sleep(3)
                    image[CamIdx] = converter[CamIdx].Convert(grabResult) # Access the openCV image data
                    image[CamIdx] = image[CamIdx].GetArray()

                    if(image[CamIdx] is not []): # !=
                        image[CamIdx] = cv2.cvtColor(image[CamIdx], cv2.COLOR_BGR2RGB)
                        image[CamIdx] = cv2.resize(image[CamIdx], (1024,780))
                        print(image[0][550, 750, 0])
                        image[CamIdx] = ImageTk.PhotoImage(image = Image.fromarray(image[CamIdx]))
                        
                        if panel[CamIdx] is None:
                            panel[CamIdx] = tk.Label(image=image[CamIdx])
                            panel[CamIdx].image = image[CamIdx]
                            panel[CamIdx].pack(side="left")
                            #panel[CamIdx].place(x=(345*CamIdx)+(20*CamIdx)+20, y=100)
                            panel[CamIdx].place(x=(1024*CamIdx)+(20*CamIdx)+20, y=100)
                        else:
                            panel[CamIdx].configure(image=image[CamIdx])
                            panel[CamIdx].image = image[CamIdx]
                else:
                    print("Error: ", grabResult.ErrorCode)
        
                grabResult.Release()

        except genicam.GenericException as e:
            # Error handling
            print("An exception occurred.", e.GetDescription())  
```
When I print the first frame of camera 1 in line ` print(image[0][550, 750, 0])` I find that there are two values prited as `98 255`. What I expect to see when the LED is ON, is only `255` and this situation happens for the first 2 to 4 frames during a time sleep of `3 seconds`

1. In order to solve this how can I skip the first 4 t0 10 frames for Basler camera in python?
2. what should I change in my function such that I can avoid multiple grabbing because I am expecting that when I will add camera 3 then I will expect to have 3 values in the mentioned above `print line (print(image[0][550, 750, 0])`

Thanking you in advance

thiesmoeller, 2021-12-16:
In a multithreaded application I would really recommend 
a) to avoid globals without proper locking
b) if you print something from a thread it is really helpful to know from which thread this info originates.

```
import threading
import logging
import numpy as np

logging.basicConfig(level=logging.DEBUG, format='%(threadName)-4s: %(message)s', )

def LiveThread():
    mat = np.random.random((100, 100))
    logging.debug(f'mat {mat.shape} {mat}'.replace("\n"," "))

cam_threads = []
for idx in range(10):
    cam_threads.append(threading.Thread(name=f"cam{idx}", target=LiveThread))

for t in cam_threads:
    t.start()

```

Jondila, 2022-12-16:
Sorry for writing here! I know it is not proper. Does the Basler camera have some sample codes in java? (reading frames and writing frames)