basler/pypylon #443: Only one camera available

jellyproll, 2021-12-09:
Hello, i have a problem with pylon (maybe somebody know solution). I have 3 basler cameras and 3 network cards in one pc. When i try connect cameras to pc pylon shows me what he see 3 cameras, but can connect only to one (with ip all is ok). If i pull out cable with "ok" camera then pylon change status to "ok" in one of remaining cameras. What i need to do to get all 3 work cameras?
[image]

thiesmoeller, 2021-12-10:
Assign separate networks to each of your interfaces.
or
set up a bridge with all your interfaces

At the moment your interfaces run in LLA mode (169.254.x.x), thus in the same network.
This doesn't work.

jellyproll, 2021-12-10:
@thiesmoeller For separate networks same problem, doesn't work 3 cameras

[image]

thiesmoeller, 2021-12-10:
you can't set 169.254.x.x to a subnet mask of /24  this is a predefined /16 network.
You have to assign three different networks to your three cards.
e.g.
host: 192.168.1.1/24 camera host: 192.168.1.2/24
host: 192.168.2.1/24 camera host: 192.168.2.2/24
host: 192.168.3.1/24 camera host: 192.168.3.2/24

jellyproll, 2021-12-10:
It's works, thank you very much!

closed 2021-12-10 by jellyproll