basler/pypylon #444: EnumerateDevices returns empty list after the computer reboots

maxwxzheng, 2021-12-14:
I'm using the following code to find all usb3 cameras connected to the system.
```
transport_layer = pylon.TlFactory.GetInstance()
devices = transport_layer.EnumerateDevices()
```
After rebooting the system, the first time we open our program and execute this code, we always get an empty list. If we reopen our program and execute this code again, we will get the list of devices. It seems very reproducible.
Any ideas on why this is the case?

thiesmoeller, 2021-12-15:
System? 
OS ?
docker?

As pylon is relying on device enumeration of the USB subsystem of your OS:
what is the time between usb-subsystem enumerating the devices and the start of your program?

maxwxzheng, 2021-12-15:
It's running on Ubuntu 18.04. We are not using docker.
`what is the time between usb-subsystem enumerating the devices and the start of your program?`
I'm not sure about this. When does the usb-subsystem enumerates the devices? Our program is used as a desktop application.

thiesmoeller, 2021-12-16:
>When does the usb-subsystem enumerates the devices? Our program is used as a desktop application.
a) can you output `lsusb` before you start your desktop application?
b) and .... you can run `tlf.EnumerateDevices()` in a loop until your devices appear

maxwxzheng, 2021-12-22:
@thiesmoeller Thanks for the response. Retrying worked. Although it's quite indeterministic. Sometimes it needs to retry 3 times. Sometimes it doesn't need retry at all. Do you know the root cause of this type of issue?