basler/pypylon #446: Basler acA2500-14gc (23363392) Image acquisition on "Basler acA2500-14gc (23363392)" failed! Error: 0xe1000014 "The buffer was incompletely grabbed.

husnejahan, 2021-12-17:
Basler acA2500-14gc (23363392) Image acquisition on "Basler acA2500-14gc (23363392)" failed! Error: 0xe1000014 "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size."

thiesmoeller, 2021-12-18:
Have you already followed the recommendations to fix this issue in the error message that you posted ? ( Inter packet delay )

husnejahan, 2021-12-22:
Yes, although I am getting the following error:

Error: 3774873620 - The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.

thiesmoeller, 2021-12-22:
In the pylon viewer "Tools" Menu is the entry "Bandwidth" manager.
If you are using a current version of pylon, there will be a "optimize" option, that automatically will find settings that give a correct starting point.

See also https://docs.baslerweb.com/bandwidth-manager

You should be able to find a working solution by increasing inter-packet-delay.

On some systems reducing PacketSize could also help ( e.g. 9000 Byte announced, but not really supported by the HW )
1500 Byte is guaranteed on all systems.

SMA2016a, 2021-12-23:
as you can see in pylon viewer, the system  is losing all frames while using the packet size 9000. So I safely assume that your Network card isn't set to support jumbo frames.
what is your operating system? if Linux have a look in network meetings if MTU size also set to 9000?

kesaroid, 2023-12-11:
Hi, the network adapter on my system supports Jumbo frames, and is set to 9014 bytes. When I optimize using bandwidth manager, it sets packet size to the highest value allowed by the camera 8192, and inter-packet delay to a whopping 37035. 
But I have about 14 cameras and see a lot of frames dropped
[image]

SMA2016a, 2023-12-12:
Firstly, I'd like to highlight that the root cause of your system issue might not be directly linked to this particular case. The error code remains consistent.

It would be advisable to open a new case and include comprehensive details about all hardware components involved, such as the network switch, the chipset of your network controller, current CPU load, etc.

clausMeko, 2024-09-23:
I do have the same issue as @SMA2016a.

Also set MTU to 9000. I have something that might help to fix it.

@thiesmoeller 
In the pythonviewer settings:
* **"Stream Parameters" > "Socket Buffer"**
how can I change this value?

I have a notebook where grabbing works (the parameter is set to 4096). On my IPC it is only 208. I cannot edit this value (neither does the gigE too not the Bandwith Optimization tool) I suspect this value is somehow derived from the System memory.

Could you please share any insight?

clausMeko, 2024-09-23:
@SMA2016a 

Would you please try if [this](https://www.baeldung.com/linux/udp-socket-buffer#resizing-the-udp-buffer) fixes your problem as well?