basler/pypylon #447: Help with hardware triggering multiple cameras simultaneously.

rafwaf, 2021-12-18:
Hi I am trying to trigger multiple cameras from a hardware signal. I'm triggering both cameras an arbitrary amount of times, and the hardware triggers can come at unexpected times (within a certain timeout window), and the cameras are triggered in an unexpected order.
My latest attempt doesn't seem to do anything.
Any help would be appreciated.

`from pypylon import pylon
from pypylon import genicam
import sys

countOfImagesToGrab = 20 # Number of images to be grabbed
maxCamerasToUse = 2 # Limits the amount of cameras used for grabbing
timeout = 20000 #max time to wait for grabbed image
serial_numbers = ['40087044']

# The exit code of the sample application.
exitCode = 0

try:
    
    tlFactory = pylon.TlFactory.GetInstance()
    
    # get all attached devices
    devices = tlFactory.EnumerateDevices()
    
    # look for unique devices that match the serial numbers
    camera_devices = []
    for device in devices:
        serial = device.GetSerialNumber()
        name = device.GetModelName()
        print(serial, name)
#        for sn in serial_numbers:
#            if serial == sn:
#                camera_devices.append(device)   
#                serial_numbers.remove(serial)
#                break                        
              
    camera_devices = tuple(devices)        #(camera_devices)
    
    # exit if no camera found
    if len(camera_devices) == 0:
        raise pylon.RuntimeException("No camera present.")
        
    print(str(len(camera_devices)) + ' cameras found.')

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cam_array = pylon.InstantCameraArray(min(len(camera_devices), maxCamerasToUse))
        
    for i, camera in enumerate(cam_array):
        
        # Create an instant camera object
        camera.Attach(tlFactory.CreateDevice(camera_devices[i]))
    
        #camera name
        camera_name = '_'.join([camera.GetDeviceInfo().GetModelName(),
                                camera.GetDeviceInfo().GetSerialNumber()])
      
        camera.Open()

        # set camera parameters
        camera.MaxNumBuffer = 5 #count of buffers allocated for grabbing
        camera.TriggerSelector.SetValue('FrameStart')
        camera.TriggerMode.SetValue('On') #hardware trigger
        camera.TriggerSource.SetValue('Line1')
        camera.TriggerActivation.SetValue('RisingEdge')
        camera.StartGrabbingMax(countOfImagesToGrab)
#        camera.ExposureAuto.SetValue('Off')
#        camera.ExposureTime.SetValue(200)
        
        # Print the model name of the camera.
        print("Setup of camera ", camera_name, " complete.")
        
except genicam.GenericException as e:
    for camera in cam_array:
        camera.Close()
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1       

converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_RGB8packed #BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

tiles = []
try:  
#    for i, camera in enumerate(cam_array):
    while cam_array.IsGrabbing():
            
            # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
        grabResult = cam_array.RetrieveResult(timeout, pylon.TimeoutHandling_ThrowException)
        print(grabResult.GetCameraContext())
        grabResult = converter.Convert(grabResult)
#            print("GrabSucceeded: ", grabResult.GrabSucceeded())
        print(grabResult.GetArray()[:1,:1])
        print(grabResult.GetArray().shape)
        tiles.append(grabResult.GetArray())
        grabResult.Release()
    cam_array.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e.GetDescription())
    exitCode = 1
finally:
    for camera in cam_array:
        cam_array.Close()

`

thiesmoeller, 2021-12-19:
You start the grabbing individually in first for loop but wait globally in while loop

Please check our extended examples in https://github.com/basler/pypylon-samples

zoldaten, 2021-12-22:
Is it possible to capture from 2 cameras synchronously without hardware trigger ?

I crashes with error:
Traceback (most recent call last):
  File "/home/pi/Desktop/.../1.py", line 38, in <module>
    grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
  File "/home/pi/.local/lib/python3.7/site-packages/pypylon/pylon.py", line 4624, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
_genicam.GenericException: QueueBuffer() failed: Device has been removed from the PC. : RuntimeException thrown (file 'PylonUsbStream.cpp', line 371)

Perhaps, 2 cameras can work in threads ?

thiesmoeller, 2021-12-22:
@zoldaten : please open a new ticket as #447 is about _using_ HW trigger....

rafwaf, 2021-12-22:
Thanks @thiesmoeller . I'll try starting the grab globally. Do I need to set the HW triggers individually or globally?

thiesmoeller, 2021-12-22:
They have to be set individually.
Only start/stop/open/close/retrieveresult .. work globally

rafwaf, 2021-12-22:
Thanks @thiesmoeller for the prompt reply. 
By the way, I was not aware of the pypylon samples repo, only the samples folder. It was nice to learn about it from your last reply. (Perhaps it can be featured more prominently in the readme of the main pypylon repo).

aruna-ram, 2022-07-14:
Hi @rafwaf, 

In your code above, do you manage to retrieve synchronised frames from the two cameras? When I use `cam_array.RetrieveResult` then I only get one image from one camera and subsequent images were not synchronised. 
If you had any example code of obtaining images from two hardware triggered cameras that would be exceptionally useful too! 
Thanks :)

seakkas, 2023-02-08:
I am also facing not synchronized grabs. I want to stitch images from two cameras. I created two queues to store grabbed images and stitch them. But I get more images from one of the cameras after a while. I use a hardware trigger that is connected to both cameras. Any suggestions?