basler/pypylon #448: fail with capture from multiple usb3.0 acA4112-20um

zoldaten, 2021-12-22:
I am using example code from here - https://github.com/basler/pypylon-samples/blob/main/notebooks/multicamera_handling.ipynb to start capture with 2 cameras in parallel. 
Both cameras connected to 2 usb3.0 ports on raspberry pi4 (Raspbian Buster).
My code almost the same:
```
import pypylon.pylon as pylon
import numpy as np
#import matplotlib.pyplot as plt
import cv2
import os

NUM_CAMERAS=2
tlFactory = pylon.TlFactory.GetInstance()
camera_devices = tuple(tlFactory.EnumerateDevices())#2
maxCamerasToUse = 2
print(str(len(camera_devices)) + ' cameras found.')
#cam_array = pylon.InstantCameraArray(min(len(camera_devices), maxCamerasToUse))
cam_array = pylon.InstantCameraArray(2)
 
for idx, cam in enumerate(cam_array):
    cam.Attach(tlFactory.CreateDevice(camera_devices[idx]))
cam_array.Open()
for idx, cam in enumerate(cam_array):
    camera_serial = cam.DeviceInfo.GetSerialNumber()
    print(f"set context {idx} for camera {camera_serial}")
    cam.SetCameraContext(idx)
# set the exposure time for each camera
for idx, cam in enumerate(cam_array):
    camera_serial = cam.DeviceInfo.GetSerialNumber()
    print(f"set Exposuretime {idx} for camera {camera_serial}")
    #cam.ExposureTimeRaw = 10000
    cam.ExposureTime = 30000
#wait for all cameras to grab 10 frames
frames_to_grab = 10
# store last framecount in array
frame_counts = [0]*NUM_CAMERAS
cam_array.StartGrabbing()
while True:
    with cam_array.RetrieveResult(1000) as res:
        if res.GrabSucceeded():
            img_nr = res.ImageNumber
            cam_id = res.GetCameraContext()
            frame_counts[cam_id] = img_nr
            print(f"cam #{cam_id}  image #{img_nr}")
            
            # do something with the image ....
            
            # check if all cameras have reached 100 images
            if min(frame_counts) >= frames_to_grab:
                print( f"all cameras have acquired {frames_to_grab} frames")
                break
            
cam_array.StopGrabbing()
cam_array.Close()
print(frame_counts)
```
But when i start it crashes:
```
2 cameras found.
set context 0 for camera 40035510
set context 1 for camera 40035511
set Exposuretime 0 for camera 40035510
set Exposuretime 1 for camera 40035511
Traceback (most recent call last):
  File "/home/pi/Desktop/.../3.py", line 32, in <module>
    cam_array.StartGrabbing()
  File "/home/pi/.local/lib/python3.7/site-packages/pypylon/pylon.py", line 4570, in StartGrabbing
    return _pylon.InstantCameraArray_StartGrabbing(self, *args)
_genicam.RuntimeException: PrepareGrab failed for device '2676:ba02:2:1:13'. Error: 'The requested operation timed out.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 262)
```
How deal with that ?

thiesmoeller, 2021-12-22:
have you applied the recommendations from the pylon installer package?

```
USB Vision Cameras
------------------
* To get the best perfomance, you should use the
  <path to pylon>/share/pylon/setup-usb.sh script, which will increase the USB
  kernel space and the file handle limit. Moreover, it will install the udev
  rule to allow access to U3V cameras for all users.
  If you want to make these system changes yourself, refer to next sections on
  how to do this.

* Increasing Packet Size
  For faster USB transfers, you should increase the packet size. If you're
  working with the pylon Viewer, you can do this by changing the
  "Stream Parameters" -> "Maximum Transfer Size" value from inside the
  pylon Viewer. If you're writing your own application, use the camera API
  to set the "Maximum Transfer Size".
  After increasing the package size, you will likely run out of kernel space
  and see corresponding error messages in the console. The default value set
  by the kernel is 16 MB. To change the value (in this example to 1000 MB), you can
  execute as root:
    sh -c 'echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb'
  This would assign a maximum of 1000 MB to the USB stack.

* Increasing File Handle Limit
  For using multiple cameras, it may be necessary to raise the handle limit.
  Typically, this is configured to 1024 handles per process. Raise this limit
  using the ulimit command, e.g.,
  ulimit -Sn 4096
  This would set the limit for the process to 4096 file handles.
  To change the settings permanently on a ubuntu system, edit
  /etc/security/limits.conf and add the line:
    *    hard    nofile    4096

* Adapt the Real-Time Attributes
  The USB module uxapi tries to raise the priority of critical threads above
  normal. This requires special permissions. For further infomation, see the
  "Permissions for Real-Time Thread Priorities" section below.

```

To persistently set this see e.g. : https://raspberrypi.stackexchange.com/questions/105800/persistently-increase-usbfs-memory-mb-on-rpi4-running-ubuntu-19

zoldaten, 2021-12-24:
Ok. It`s seems to me it is working. But a bit unstable. After about 7300 iteration it crashes:
```
Traceback (most recent call last):
  File "/home/pi/Desktop/.../test_basler3--.py", line 52, in <module>
    with cam_array.RetrieveResult(1000) as res:
  File "/home/pi/.local/lib/python3.7/site-packages/pypylon/pylon.py", line 4624, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
_genicam.GenericException: QueueBuffer() failed: Device has been removed from the PC. : RuntimeException thrown (file 'PylonUsbStream.cpp', line 371)
```
What i`ve done:
$ sudo sh -c 'echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb'
$ ulimit -Sn 4096
Set cameras parameters with api. At start they are:
```
2 cameras found.
W: 600 H: 1440 fps: 20.0 Exp: 30000.0 MaxNumBuffer: 30 AcquisitionFrameRate: 20.0
BslUSBSpeedMode: SuperSpeed DeviceLinkSpeed: 500000000 DeviceLinkCurrentThroughput: 17281680 DeviceLinkThroughputLimit: 360000000 DeviceLinkThroughputLimitMode: Off
W: 600 H: 1440 fps: 20.0 Exp: 30000.0 MaxNumBuffer: 30 AcquisitionFrameRate: 20.0
BslUSBSpeedMode: SuperSpeed DeviceLinkSpeed: 500000000 DeviceLinkCurrentThroughput: 17281680 DeviceLinkThroughputLimit: 360000000 DeviceLinkThroughputLimitMode: Off
```
And the grab code is the same only counter is added:
```
while True:
    with cam_array.RetrieveResult(1000) as res:
        if res.GrabSucceeded():
            img_nr = res.ImageNumber
            cam_id = res.GetCameraContext()
            frame_counts[cam_id] = img_nr            
            print(counter)
            counter+=1
```
Are there any more tricks ? May be something like "usb flush" or "grabrelease" after each frame? 
I saw example about using Zerocopy. Is it suitable in here?

p.s.
```<path to pylon>/share/pylon/setup-usb.sh``` can not find it on raspberry. i see only pylon viewer installed.
``` Adapt the Real-Time Attributes``` - can`t catch how to do that ?  The link you kindly provide is
about increase of usb limits.

thiesmoeller, 2022-01-05:
You still have to set the transfersize in your program.
Otherwise it is running with system defaults, which are too low.
before starting the stream use for a single camera e.g.:

`cam.StreamGrabber.MaxTransferSize = 4 * 1024 * 1024`

zoldaten, 2022-01-11:
crashes (( : 
'''
Traceback (most recent call last):
  File "/home/pi/Desktop/.../test_basler3.py", line 101, in <module>
    grabResult = camera.RetrieveResult(200, pylon.TimeoutHandling_ThrowException)
  File "/home/pi/.local/lib/python3.7/site-packages/pypylon/pylon.py", line 3502, in RetrieveResult
    return _pylon.InstantCamera_RetrieveResult(self, *args)
_genicam.GenericException: Failed to register buffer at ux stream for device '2676:ba02:2:2:44'. Error: 'Device has been removed from the PC.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 209)
'''
p.s. changing 200 to 5000 (does not make an effect)