basler/pypylon #449: pypylon build error

hlacik, 2021-12-24:
when running :

`/opt/pypy/bin/pypy3 -mpip install git+https://github.com/basler/pypylon@1.7.4#egg=pypylon`

```
    /tmp/pip-install-3sdw039m/pypylon_a518166c7d114021990d3e579fe2cc3e/generated/pylon_wrap.cpp: In function 'int Pylon_CGrabResultData_GetNumBufferExports(Pylon::CGrabResultData*, PyObject*)':
    /tmp/pip-install-3sdw039m/pypylon_a518166c7d114021990d3e579fe2cc3e/generated/pylon_wrap.cpp:4257:29: error: 'struct PyMemoryViewObject' has no member named 'mbuf'
``

```
root@7e9f7490a667:/pylon_setup# /opt/pypy/bin/pypy3 --version                                                            
Python 3.8.12 (9ef55f6fc369, Oct 25 2021, 13:25:27)
[PyPy 7.3.7 with GCC 9.3.1 20200408 (Red Hat 9.3.1-2)]
```

i know pypy is not supported yet, but i am running all my workloads using pypy so really curious how to solve this

thiesmoeller, 2021-12-24:
Never tried ...
First check that swig version is compatible with  pypy

hlacik, 2021-12-25:
I am using latest swig 4.0.2 , with latest pypy - 3.8 prebuilt binary on https://www.pypy.org/download.html .

Could you please give it a try? So far I have switched on your all python projects to pypy and performance increase is incredible, this is the last repository we can not due to this problem - mbuf is not known to PyMemoryViewObject

thiesmoeller, 2021-12-26:
can you test in your environment: `pypy3 -mpip install git+https://github.com/thiesmoeller/pypylon@fix/pypy_compat_fix#egg=pypylon`
I excluded the zero copy support, which is currently the reason for pypy not compiling..
... and not mandatory to run pypylon