basler/pypylon #450: pylon IP Configurator on Ubuntu 18.04 cannot find Basler acA2500-14gm GigE camera

maxwxzheng, 2021-12-27:
I'm having trouble finding the camera with the ip configurator. Is there anything else I need to check?

OS: Ubuntu 18.04
Camera: Basler acA2500-14gm GigE
Network adapter: 1.25G Gigabit Ethernet Server Network Adapter(NIC) with Intel I210 Controller
Firewall: off

thiesmoeller, 2021-12-29:
a) Power supply to camera active 12V or POE?
b) have you checked the readme of the pylon install /opt/pylon/share/pylon/README or in the installer tar file
   -> * Problem: My application can't see my camera. I'm not able to
  reconfigure the camera to make it visible again.

  Solution:
  First make sure you don't have a firewall enabled on the network
  adapter the camera is connected to.
  If you still can't see the camera, reverse-path filtering in the kernel may
  prevent your application from detecting the camera. This can happen if the 
  camera's IP address is not within the same subnet as the network adapter 
  the camera is attached to.
  To check whether filtering is turned on, run the following command:
    sysctl -a 2>/dev/null | grep '\.rp_filter'

  In the output look for the following lines:
   net.ipv4.conf.all.rp_filter=1
   net.ipv4.conf.eth1.rp_filter=1

  where "eth1" is the network adapter the camera is connected to.
  The "net.ipv4.conf.all.rp_filter" is a global switch that must be turned off.
  The "net.ipv4.conf.eth1.rp_filter" shows whether filtering for the specified
  network adapter is enabled. To disable filtering, you must first turn off
  filtering for "all" and the specific network adapter (in this
  sample "eth1"). Use the following commands to change the filtering behavior
  at runtime:
    sudo sysctl net.ipv4.conf.all.rp_filter=0
    sudo sysctl net.ipv4.conf.eth1.rp_filter=0

  Restart your application and check whether the camera(s) are
  detected. Reconfigure the camera(s) and use "Write Configuration" to make
  your changes persistent. You can re-enable filtering by executing the same
  commands but set a value of 1. If you want to turn off filtering permanently,
  you can edit the same values in /etc/sysctl.conf.

  From Linux kernel version 2.6.32 onwards, the rp_filter settings allow strict
  and loose filtering. To accept asymmetrically routed packets, modify /etc/sysctl.conf:
    net.ipv4.conf.default.rp_filter = 2
    net.ipv4.conf.all.rp_filter = 2

maxwxzheng, 2022-01-30:
Turned out to be hardware issue. The network adapter I got was not compatible with ubuntu 18.04. Switched to another network adapter and it's all good.

closed 2022-01-30 by maxwxzheng