basler/pypylon #454: Camera does not immediately accept new image format settings when grabbing images

mjr1ch, 2022-01-06:
Hi 

I have a Basler acA4024-29um camera that I am using in a CV application. I am trying to process and save images at the ~ 30 fps offered by the camera. To aid in that endeavor I crop the field of view to cut down on the amount of pixels I am looking at, so I take advantage of the offset, height and width parameters exposed by the API. 

My problem is after setting the appropriate values and then initiating continuous image aquisition what I noticed is I get 2-3 seconds of larger images reflecting the default size values. I would expect that since I changed everything before image grabbing started that I should not get one image at the default values. 

I think this is a bug or am I missing something. 

Thanks in advance.

thiesmoeller, 2022-01-06:
As getting images with an ROI different from what  you have configured _before_ StartStreaming is not possible by design,
( unless you would be using the sequencer control feature ) 
it would really help if can post a small sample code, that demonstrates the issue.

mjr1ch, 2022-01-14:
So I found one bug in that I had an internal queue that had to be cleaned up but I still have the issue with the camera from time to time (usually when I am running something important ) that would throw an error and mess up my run. This only started happening when I started playing with the image format settings from python. 

When it does fail I only get two images of differing sizes. The 2nd image is always smaller than the first. I cannot predictably reproduce the error though. I saw something similar with #255. I tried something similar of cleaning the buffer after stop grabbing. we shall see how that goes. 

my code that I am using currently.... 

`# A thread that produces data
def image_producer(camera, out_q, running ):
    print('formatt0ing image')
    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
    counter = 0 
    # Produce some data
    print('launching camera thread')
    while (camera.IsGrabbing() and running):
        #grabResult = self.loop.run_in_executor(self.pool,self.camera.RetrieveResult,5000, pylon.TimeoutHandling_ThrowException)
        try:
            grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                img = converter.Convert(grabResult)
                img = img.GetArray()
                out_q.put(img)
                grabResult.Release() 
                counter = 0 
            print(f'\r Running signal {running} ', end = '', flush = True)
        except:
            counter  += 1
            print(f'\rCamera stalled {counter} times', end = '', flush = True)
            break
    print('stop signal recieved')
    camera.StopGrabbing()
    return `