basler/pypylon #458: The max fps for software trigger (Action Command)

yctsai-metgauge, 2022-01-11:
Hi
I would like to take pictures in a high rate (40hz) synchronously on 2 cameras (will up to 12 cameras in the future).
I am using 2 acA3800-10gm camera, I use action command to take pictures synchronously.
My ROI is 256*256. According to frame rate calculator, the theoretical maximum frame rate is 79.

I issue action command with time.sleep to get the target hz I want by following code:
```
for i in range(400):
    print('issuing no-wait action command')
    act_cmd.IssueNoWait()
    time.sleep(25/1000)  # 25 ms between each action command
```

I try to save pictures of 2 cameras by following code:
```
cam_index=[0,0]
cam_array.StartGrabbing()

try:
    while cam_array.IsGrabbing():
        grabResult = cam_array.RetrieveResult(10000 ,pylon.TimeoutHandling_Return)
        if grabResult.GrabSucceeded():
            img.AttachGrabResultBuffer(grabResult)
            cameraContextValue = grabResult.GetCameraContext()
            filename = f"cam{cameraContextValue}_img{cam_index[cameraContextValue]}.png" 
            cam_index[cameraContextValue]+=1
            img.Save(pylon.ImageFileFormat_Png, filename)
except:
    cam_array.StopGrabbing()
    cam_array.Close()
```
---
I target my camera to stopwatch to get the timestamp. 
The 3 digits shows the millisecond in a second. 
I found many duplicate pictures in my saving pictures. 

[image]
---
Here, I list some possible reason to explain that
1. The python code is not efficient, the process of saving pictures is stuck, I should use c++ code.
2. The limit of software trigger is slower than the speed of sending action command, I should use a hardware trigger instead.
3. The bandwidth does not allow me to send the pictures in such speed (less possible, cuz I am using a 1 Gigabit Ethernet network and the ROI is only 256*256).

Please let me know if you have any idea.

SMA2016a, 2022-01-22:
1) I think that the point one isn't the root cause.
2) I don't think
3) I don't think

I think frequency of the monitor  and the rolling shutter effect.

yctsai-metgauge, 2022-02-22:
Thanks to @SMA2016a 
I think the problem is on the frequency of the monitor. I then use a cell phone app to do the same test and get a desirable result.
For someone who would like to do the same test:
Although both my monitor and my cell phone (google pixel 4a) claim that the refresh rate is 60 hz, the monitor actually refresh in a slower rate according to my experiment.
So, skeptical about everything in experiment.

closed 2022-02-22 by yctsai-metgauge