basler/pypylon #459: Difference between pylon viewer and pypylon

Drulludanni, 2022-01-14:
Hi I'm running into an issue where I get a difference in output between the pylon viewer and the pylon library in python even though the settings are set to exactly the same.  So when I take a picture using the pylon viewer I get the picture with the quality that I'd like, when I try to take the same picture using pypylon I get back a monchrome image with a weird dotted pattern, so I pass it to an `ImageFormatConverter()` with `SetOutputPixelFormat = pylon.PixelType_RGB8packed` and instead get an output image that does not have the pattern and has the image in acceptable quality except it is still monochrome. So then I tried taking the picture by changing the setting` PixelFormat` to `RGB8` which gives me a color image, however it has significantly more artifacts than the image in pylon viewer captured with the pixel format as Bayer RG 8 which has the quality I desire (using RGB8 produces the same artifacts).

what process is going on that allows pylon viewer to get a higher quality colored images than pylon? How can I replicate that?

just to illustrate the difference I've uploaded these 2 images taken with the different pixel formats, if you observe the numbers in the image it is clear that there is a lot of yellow artifacts in the RGB8 image that are not existent in the RG8 image.

[image]
[image]

Drulludanni, 2022-01-18:
Ok, so I managed to figure out what was wrong, it was a simple spelling error in my code `converter.SetOutputPixelFormat = pylon.PixelType_RGB8packed` should have been `converter.OutputPixelFormat = pylon.PixelType_RGB8packed`, it did not crash which was weird, but behaved incorrectly

closed 2022-01-18 by Drulludanni

thiesmoeller, 2022-01-18:
This is a current  bug in pypylon

Please use the string versions of the names.

So "RGB8Packed" not the integer definition value