basler/pypylon #460: how to parse TimeStamp using cameras ?

zoldaten, 2022-01-14:
i`m trying grabmultiplecameras.py from repo samples.
Added to code: time_stamp=res.TimeStamp (as i saw res has it).
See something like this:
```
cam #0  image #1 time #1887938185770
cam #1  image #1 time #5674457758239
cam #0  image #2 time #1887980750406
cam #1  image #2 time #5674500322875
```
How to parse such TimeStamp ? How to set up cameras to use right time ?

thiesmoeller, 2022-01-14:
Which Interface?

USB one tick is 1e-9s counts from start of device
GEV standard one tick is 8e-9s on gigabit speed from start to of device or timestampreset
GEV + IEEE1588 is 1e-9 in time of clock master.
If you have a clock master that is e.g. gps synchronized, timestamps are precise UTC

zoldaten, 2022-01-15:
i use usb cameras and RTC-hardware clock on raspberry pi 4. 
May be it can be copy from hw-clock to Timestamp at start ? 

And how to read time like 1887938185770 ? 
Something like ```print(datetime.utcfromtimestamp('1887938185770').strftime('%Y-%m-%d %H:%M:%S')) ``` doesnt work.

PolymaTh-EE98, 2022-01-15:
try this

import time
print(time.strftime('%Y-%m-%d %H:%M:%S',  time.localtime(time.time())))
time.time() is in seconds so for the camera time try
print(time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(ImageTimeStamp_out / 1000000))))

thiesmoeller, 2022-01-15:
@PolymaTh-EE98 
This would only work for GEV +1588

@zoldaten is using a USB device

Syncing to your host is not possible.
Typical solution is to match the image timestamp to system time on receive.

zoldaten, 2022-01-15:
Sad news (
As far as i found camarray is not suitable if one`s want to use it with usb cameras and do it real-time:
```
2 cameras found.
set context 0 for camera 40035510
set context 1 for camera 40035511
set Exposuretime 0 for camera 40035510
set Exposuretime 1 for camera 40035511
cam #0  image #1 time #1139333657787
cam #1  image #1 time #1138105999971
cam #0  image #2 time #1139376222423
cam #1  image #2 time #1138148564607
all cameras have acquired 2 frames
[2, 2]

>>> print(time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(1139333657787 / 1000000)))
1970-01-14 07:28:53
>>> print(time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(1139376222423 / 1000000)))
1970-01-14 07:29:36
>>> print(time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(1138105999971 / 1000000)))
1970-01-14 07:08:25
>>> print(time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(1138148564607 / 1000000)))
1970-01-14 07:09:08
```
If put away that the start date is always random time latensy is too high.

Tested on Odroid XU4 with 2 usb cameras. (Raspberry pi 4 cant deal with them (800x2000 resolution) because of usb trasmission limits).

Thanks for help!

thiesmoeller, 2022-01-15:
The timestamp of the cameras are very precise ( ~50ppm ) , 
although not synchronized to other cameras in the same system on USB as explained.

What do want to reach with the two cameras? precise stereo capturing?
In this case really recommend to use hardware trigger of the cameras

InstantCameraArray is capable of doing real time reception of data, as the underlying grab-engines all work in parallel.
So you only get the images sequentially, but the original acquisition in the cameras could be in sync to microsecond precision.

zoldaten, 2022-01-16:
Actually i work on 2 questions:
1. How to synchronize 12 usb cameras and 1 encoder. As i have 12 raspberry pi each of them uses  only 1 camera and code:
```
if grabResult.GrabSucceeded():
        image = converter.Convert(grabResult)
        img = image.GetArray()
        #cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        #cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
        if decode_my(img)!=None:
                a=decode_my(img)               
                with open("output.csv", "a") as csv:
                    csv.write("{} {} {}\n".format(datetime.datetime.now().strftime('%d/%m/%y %H:%M:%S:%f'),a,"cam1"))                
                    csv.flush()
                counter+=1
                image.Release()
```
2. Is it possible to use 2 usb cameras on 1 raspberry pi (odroid, etc.) ?

thiesmoeller, 2022-02-15:
Hi @zoldaten, 

issue still open ? ... as you reopened?

zoldaten, 2022-02-15:
could you show the way how to synchronize 12 usb cameras ? i use timestamps for now to find delta of screenshots.

thiesmoeller, 2022-02-15:
You wrote, that you have an encoder. So HW trigger of all cameras is the best solution

zoldaten, 2022-02-15:
ok, i`ll try to glue it up.

closed 2022-02-15 by zoldaten