basler/pypylon #463: General support for GenICam-compliant Cameras

agvdndor, 2022-01-27:
Hi,

Basler cameras support The GenICam standard and provide GenTL producers so that they can be operated by third-party software libraries that support the GenTL standard, i.e. GenTL consumers. It is my understanding that (py)pylon itself is such a GenTL consumer and hence should be able to also operate other, non-basler, GenTL producers. 

I have tested this theory by enumerating devices after adding the simulated GenTL producer `TLSimu.cti` (provided by the [official genicam python bindings](https://pypi.org/project/genicam/))  to the `GENICAM_GENTL64_PATH` system variable. The simulated device does not show up in the device_list. 

Was I wrong in assuming that pypylon uses the GenTL interface to operate its cameras?

SMA2016a, 2022-01-27:
pylon does not support 3rd party cameras, unless you get an unlock key from Basler.

agvdndor, 2022-01-27:
Thanks for the reply. Could you elaborate on that? 
1. How would one go about getting such a key? I can't seem to find any documentation on it.
2. What exactly is its purpose? Are 3rd party GenTL producers now simply restricted as a business decision?

SMA2016a, 2022-01-28:
Simply contact the Basler  sale rep in your region.

https://www.baslerweb.com/de/vertrieb-support/vertrieb/