basler/pypylon #464: Make Exe File of Pypylon Project

sarperkilic, 2022-01-31:
Hello,

I have developed a python project using pypylon library and then I am trying to conver this python script to exe file.  But, after creating an executable file I get errors when I try to run it. It is not working. 

When I delete the codes related to the pypylon library from my project and create the exe file again, I can open the exe file. I am pretty sure this problem is related to pypylon.

How can I convert my project to exe file correctly?

Any suggestions?

Thanks

thiesmoeller, 2022-01-31:
>trying to conver this python script to exe file

please describe the tool you use. There are multiple techonologies available to "convert to exe"

what is your:
architecture ( x86 / arm ) 
operating system ( win/linux/macos )
python version 

??

sarperkilic, 2022-01-31:
> > trying to conver this python script to exe file
> 
> please describe the tool you use. There are multiple techonologies available to "convert to exe"
> 
> what is your: architecture ( x86 / arm ) operating system ( win/linux/macos ) python version
> 
> ??

Hello,

I am using pyinstaller for converting process.

Now, I am getting this error on windows operating system, x86 architecture with pyton 3.8.

But I have to run this exe on linux also. So, could you also explain the problem for linux system please?

Thanks

thiesmoeller, 2022-02-03:
do you have the pyinstaller hooks on your system
https://github.com/pyinstaller/pyinstaller-hooks-contrib

they have integrated support for pypylon here: https://github.com/pyinstaller/pyinstaller-hooks-contrib/pull/114

sarperkilic, 2022-02-10:
Hello again,

Thanks for your help.

I have installed pypylon hook and used it as follow:

pyinstaller --onefile --paths="/usr/local/lib/python3.6/site-packages/cv2/python-3.6" --additional-hooks-dir=. main.py

Exe file was created without any problems.

But, when I try to run this exe file, I got an error I shared below.

dist ./main
Traceback (most recent call last):
  File "main.py", line 210, in <module>
  File "main.py", line 180, in setupUi
  File "take_photo.py", line 7, in __init__
  File "take_photo.py", line 11, in open_camera
  File "pypylon/pylon.py", line 1592, in CreateFirstDevice
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
[29382] Failed to execute script main

USB 3.0 camera is connected to my laptop and I can run this applicaton from python scripts. 

I am also sharing take_photo.py script below.

from pypylon import pylon
class basler:       
    def __init__(self,exposure_time) -> None:
        self.camera=self.open_camera(exposure_time)
   
    def open_camera(self, exposure_time):
        self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())  
        self.camera.Open()
        self.camera.ExposureTime.SetValue(int(exposure_time))  
        self.camera.MaxNumBuffer = 1
        return self.camera

    def take_image(self,count):        
        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_RGB8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned    
        result = self.camera.GrabOne(100)
        image = converter.Convert(result)
        result.Release()
        return image.Array

How can I solve this?

Thanks

OS: linux/ubuntu 18.04 x86
python 3.6.9

sarperkilic, 2022-02-14:
Hello again,

Is there any solution suggestion related this issue?

Thanks

> Hello again,
> 
> Thanks for your help.
> 
> I have installed pypylon hook and used it as follow:
> 
> pyinstaller --onefile --paths="/usr/local/lib/python3.6/site-packages/cv2/python-3.6" --additional-hooks-dir=. main.py
> 
> Exe file was created without any problems.
> 
> But, when I try to run this exe file, I got an error I shared below.
> 
> dist ./main Traceback (most recent call last): File "main.py", line 210, in File "main.py", line 180, in setupUi File "take_photo.py", line 7, in **init** File "take_photo.py", line 11, in open_camera File "pypylon/pylon.py", line 1592, in CreateFirstDevice _genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694) [29382] Failed to execute script main
> 
> USB 3.0 camera is connected to my laptop and I can run this applicaton from python scripts.
> 
> I am also sharing take_photo.py script below.
> 
> from pypylon import pylon class basler: def **init**(self,exposure_time) -> None: self.camera=self.open_camera(exposure_time)
> 
> ```
> def open_camera(self, exposure_time):
>     self.camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())  
>     self.camera.Open()
>     self.camera.ExposureTime.SetValue(int(exposure_time))  
>     self.camera.MaxNumBuffer = 1
>     return self.camera
> 
> def take_image(self,count):        
>     converter = pylon.ImageFormatConverter()
>     converter.OutputPixelFormat = pylon.PixelType_RGB8packed
>     converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned    
>     result = self.camera.GrabOne(100)
>     image = converter.Convert(result)
>     result.Release()
>     return image.Array
> ```
> 
> How can I solve this?
> 
> Thanks
> 
> OS: linux/ubuntu 18.04 x86 python 3.6.9

Hello again,

Is there any solution suggestion related this issue?

Thanks

thiesmoeller, 2022-02-14:
Have you checked in pyinstaller logging during "baking" of your exe, that the pylon shared libraries have been integrated?

Pylon architecture consists of a number of libraries, that are directly visible to pyinstaller dependency scanner and the transportlayer libraries which are only searched in runtime.

This should be taken care of in the hook file.

As a hint for debugging:
Don't start with the --onefile option as this will give you _no_ chance to debug it.

thiesmoeller, 2022-02-14:
Below is a simplified test script and a Dockerfile to test on 18.04 in clean environment.
This works. 

Can you do the same test on your system, to just use the CAMEMU camera, as this will prove that all libraries are found.
Just use the PYLON_CAMEMU camera?

issue_464.py:
```python
import pypylon.pylon as py

tlf = py.TlFactory.GetInstance()

cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

res = cam.GrabOne(1000)
img = res.Array

print("captured image: pixel_0,0 ", img[0,0])

cam.Close()

```

Dockerfile:
```dockerfile
FROM ubuntu:18.04
RUN apt-get update && apt-get install python3 python3-pip -y 
RUN python3 -m pip install --upgrade pip
RUN pip3 install pypylon pyinstaller numpy

COPY issue_464.py /work/issue_464.py 
RUN  pyinstaller --onefile /work/issue_464.py
# test running
RUN PYLON_CAMEMU=1 dist/issue_464

# test running without installed pypylon
RUN pip3 uninstall pypylon -y
RUN PYLON_CAMEMU=1 dist/issue_464
```

this call mis-uses docker a bit to build and test  during dockerfile build process..

` docker build -t issue_464 Dockerfile`

result:

```
Sending build context to Docker daemon  3.072kB
Step 1/9 : FROM ubuntu:18.04
 ---> dcf4d4bef137
Step 2/9 : RUN apt-get update && apt-get install python3 python3-pip -y
 ---> Using cache
 ---> e6cc2b16824d
Step 3/9 : RUN python3 -m pip install --upgrade pip
 ---> Using cache
 ---> 8160acd0ad7b
Step 4/9 : RUN pip3 install pypylon pyinstaller numpy
 ---> Using cache
 ---> 709e048bac31
Step 5/9 : COPY issue_464.py /work/issue_464.py
 ---> 8a6437221313
Step 6/9 : RUN  pyinstaller --onefile /work/issue_464.py
 ---> Running in 841fbf372a72
52 INFO: PyInstaller: 4.9
52 INFO: Python: 3.6.9
53 INFO: Platform: Linux-5.16.5-1-default-x86_64-with-Ubuntu-18.04-bionic
53 INFO: wrote /issue_464.spec
54 INFO: UPX is not available.
56 INFO: Extending PYTHONPATH with paths
[... 67 lines cut ...]
20894 INFO: Building EXE from EXE-00.toc completed successfully.
Removing intermediate container 841fbf372a72
 ---> 7b29eaece9a8
Step 7/9 : RUN PYLON_CAMEMU=1 dist/issue_464
 ---> Running in 3d4e47959eed
captured image: pixel_0,0  1
Removing intermediate container 3d4e47959eed
 ---> 9268a9bd471b
Step 8/9 : RUN pip3 uninstall pypylon -y
 ---> Running in 523414f072da
Found existing installation: pypylon 1.8.0
Uninstalling pypylon-1.8.0:
  Successfully uninstalled pypylon-1.8.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Removing intermediate container 523414f072da
 ---> 7232cfa8cb90
Step 9/9 : RUN PYLON_CAMEMU=1 dist/issue_464
 ---> Running in 270a06f42ca4
captured image: pixel_0,0  1
Removing intermediate container 270a06f42ca4
 ---> f896560958ed
Successfully built f896560958ed
Successfully tagged issue_464:latest

```

thiesmoeller, 2022-02-17:
Hello again, @sarperkilic 

Have you been able to get this running on your platform?

sarperkilic, 2022-02-18:
> Hello again, @sarperkilic
> 
> Have you been able to get this running on your platform?

Hello @thiesmoeller,

I tried the solution you suggested today. I am sharing the results with you below.

First of all, I am sharing the logs of the exe conversion process without using docker. I see there are some problems related pypylon module.

122 WARNING: Several hooks defined for module 'pypylon'. Please take care they do not conflict.
7148 INFO: Loading module hook 'hook-pypylon.py' from '/home/inovako/Documents/devs/local_lumitec/installer_deneme'...
/home/inovako/.local/lib/python3.6/site-packages/pkg_resources/__init__.py:119: PkgResourcesDeprecationWarning: 0.18ubuntu0.18.04.1 is an invalid version and will not be supported in a future release
  PkgResourcesDeprecationWarning,
7177 WARNING: Unable to copy metadata for pypylon._pylon: The 'pypylon._pylon' distribution was not found and is required by the application
7178 INFO: Determining a mapping of distributions to packages...
19148 WARNING: Unable to determine requirements for pypylon._pylon: The 'pypylon._pylon' distribution was not found and is required by the application
19149 INFO: Excluding import of pypylon._pylon from module pypylon
19150 INFO: Excluding import of pypylon._pylon from module pypylon.pylon
19150 INFO: Excluding import of pypylon._genicam from module pypylon
19150 INFO: Excluding import of pypylon._genicam from module pypylon.genicam

Then, I tried the test script and a Dockerfile you shared on my system.
I am sharing the logs with you. (I deleted the logs about installations of other packages.)

$ docker build -t issue_464 .           
Sending build context to Docker daemon  3.072kB
Step 1/9 : FROM ubuntu:18.04
18.04: Pulling from library/ubuntu
68e7bb398b9f: Pull complete 
Digest: sha256:c2aa13782650aa7ade424b12008128b60034c795f25456e8eb552d0a0f447cad
Status: Downloaded newer image for ubuntu:18.04
 ---> dcf4d4bef137
Step 2/9 : RUN apt-get update && apt-get install python3 python3-pip -y
 ---> Running in aac24749e01f
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Removing intermediate container aac24749e01f
 ---> 63c3cbcc8e1d
Step 3/9 : RUN python3 -m pip install --upgrade pip
 ---> Running in 980008175a72
Collecting pip
Successfully installed pip-21.3.1
Removing intermediate container 980008175a72
 ---> 2fcdf2014f43
Step 4/9 : RUN pip3 install pypylon pyinstaller numpy
 ---> Running in 04f6d3dbecb5
Successfully installed altgraph-0.17.2 importlib-metadata-4.8.3 numpy-1.19.5 pyinstaller-4.9 pyinstaller-hooks-contrib-2022.0 pypylon-1.8.0 typing-extensions-4.1.1 zipp-3.6.0
Removing intermediate container 04f6d3dbecb5
 ---> d2db57a8198a
Step 5/9 : COPY issue_464.py /work/issue_464.py
 ---> 699f97154182
Step 6/9 : RUN  pyinstaller --onefile /work/issue_464.py
 ---> Running in ac132ea429e2
54 INFO: PyInstaller: 4.9
54 INFO: Python: 3.6.9
55 INFO: Platform: Linux-5.4.0-99-generic-x86_64-with-Ubuntu-18.04-bionic
55 INFO: wrote /issue_464.spec
57 INFO: UPX is not available.
59 INFO: Extending PYTHONPATH with paths
['/work']
196 INFO: checking Analysis
196 INFO: Building Analysis because Analysis-00.toc is non existent
196 INFO: Initializing module dependency graph...
198 INFO: Caching module graph hooks...
207 INFO: Analyzing base_library.zip ...
3088 INFO: Caching module dependency graph...
3242 INFO: running Analysis Analysis-00.toc
[... 72 lines cut ...]
  Successfully uninstalled pypylon-1.8.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Removing intermediate container 5ade0e77ee5b
 ---> ebd62ba736b2
Step 9/9 : RUN PYLON_CAMEMU=1 dist/issue_464
 ---> Running in fc548447b709
captured image: pixel_0,0  1
Using device  Emulation
Removing intermediate container fc548447b709
 ---> c01c6c81943c
Successfully built c01c6c81943c
Successfully tagged issue_464:latest

It produced a result like you shared, so we can say it worked.
But there are still some problems.
As you can see from the docker logs, same 'pypylon distribution was not found' error occurred.

And, when I try to run the test appli
And when I tried to run the created docker image and exe the test application, I got the no device error again. The camera is connected to the system.

[image]

What should I understand from this experiment and how can I solve this no device error?

Thanks in advance, looking forward to your reply.

thiesmoeller, 2022-02-18:
Hint to something wrong on your system is:

122 WARNING: Several hooks defined for module 'pypylon'. Please take care they do not conflict.

I would always recommend to setup a virtualenv and only install the modules that you require.

Like in my example docker version

Otherwise creating the exe is not reproducible, as it depends on the current state of your system

sarperkilic, 2022-02-21:
> My experiment should demonstrate, that you can successfully package a pypylon app.
> 
> To your issue: Hint to something wrong on your system is:
> 
> 122 WARNING: Several hooks defined for module 'pypylon'. Please take care they do not conflict.
> 
> I would always recommend to setup a virtualenv and only install the modules that you require.
> 
> Like in my example docker version
> 
> Otherwise creating the exe is not reproducible, as it depends on the current state of your system

Hello @thiesmoeller again,

Today I create an environment and installed only the required module for my application.

And then, I converted the code to an exe file. I used the same code you shared with me.

basler_image.py :

`import pypylon.pylon as py

tlf = py.TlFactory.GetInstance()

cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()

res = cam.GrabOne(100)
img = res.Array

print("captured image: pixel_0,0 ", img[0,0])

cam.Close()
 `

When I run this code as follows, I got the same error:
`dist/basler_image `
Traceback (most recent call last):
  File "basler_image.py", line 5, in <module>
    cam = py.InstantCamera(tlf.CreateFirstDevice())
  File "pypylon/pylon.py", line 1601, in CreateFirstDevice
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)

With emuluate camera parameter, it works. 
PYLON_CAMEMU=1 dist/basler_image 
captured image: pixel_0,0  1

But I cannot use the exe with my USB 3.0 camera. Again, I am able to take image from connected USB3.0 camera before the exe conversion from running python script. (from terminal, python3 basler_image.py)

What should I do next? What is wrong here? I am also sharing the pyinstaller log below.

It is becoming urgent problem. Thanks for your help.

pyinstaller --additional-hooks-dir=. basler_image.py
43 INFO: PyInstaller: 4.9
43 INFO: Python: 3.6.9
43 INFO: Platform: Linux-5.4.0-99-generic-x86_64-with-Ubuntu-18.04-bionic
43 INFO: wrote /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/basler_image.spec
45 INFO: UPX is not available.
46 INFO: Extending PYTHONPATH with paths
['/home/inovako/Documents/devs/execute_conversion/venv/exec_env/test']
152 INFO: checking Analysis
152 INFO: Building Analysis because Analysis-00.toc is non existent
152 INFO: Initializing module dependency graph...
153 INFO: Caching module graph hooks...
154 WARNING: Several hooks defined for module 'pypylon'. Please take care they do not conflict.
158 INFO: Analyzing base_library.zip ...
2345 INFO: Caching module dependency graph...
2438 INFO: running Analysis Analysis-00.toc
2458 INFO: Analyzing /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/basler_image.py
[... 37 lines cut ...]
7528 WARNING: Cannot find libgfortran-ed201abd.so.3.0.0 (needed by /home/inovako/Documents/devs/execute_conversion/venv/exec_env/lib/python3.6/site-packages/numpy/core/../../numpy.libs/libopenblasp-r0-8a0c371f.3.13.so)
7537 INFO: Looking for eggs
7537 INFO: Python library not in binary dependencies. Doing additional searching...
7557 INFO: Using Python library /usr/lib/x86_64-linux-gnu/libpython3.6m.so.1.0
7562 INFO: Warnings written to /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/build/basler_image/warn-basler_image.txt
7588 INFO: Graph cross-reference written to /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/build/basler_image/xref-basler_image.html
7594 INFO: checking PYZ
7594 INFO: Building PYZ because PYZ-00.toc is non existent
7594 INFO: Building PYZ (ZlibArchive) /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/build/basler_image/PYZ-00.pyz
7985 INFO: Building PYZ (ZlibArchive) /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/build/basler_image/PYZ-00.pyz completed successfully.
7989 INFO: checking PKG
7989 INFO: Building PKG because PKG-00.toc is non existent
7989 INFO: Building PKG (CArchive) basler_image.pkg
8017 INFO: Building PKG (CArchive) basler_image.pkg completed successfully.
8017 INFO: Bootloader /home/inovako/Documents/devs/execute_conversion/venv/exec_env/lib/python3.6/site-packages/PyInstaller/bootloader/Linux-64bit-intel/run
8017 INFO: checking EXE
8018 INFO: Building EXE because EXE-00.toc is non existent
8018 INFO: Building EXE from EXE-00.toc
8018 INFO: Copying bootloader EXE to /home/inovako/Documents/devs/execute_conversion/venv/exec_env/test/build/basler_image/basler_image
8018 INFO: Appending PKG archive to custom ELF section in EXE
8024 INFO: Building EXE from EXE-00.toc completed successfully.
8025 INFO: checking COLLECT
8025 INFO: Building COLLECT because COLLECT-00.toc is non existent
8025 INFO: Building COLLECT COLLECT-00.toc
8109 INFO: Building COLLECT COLLECT-00.toc completed successfully.

thiesmoeller, 2022-02-21:
we can reproduce the issue .... it only happens with usb backend

thiesmoeller, 2022-02-21:
Hi @sarperkilic,

use this hook-file from a folder, that you reference with --additional-hooks-dir
Not beautiful but fixes  the missing .so file

```python
# ------------------------------------------------------------------
# Copyright (c) 2020 PyInstaller Development Team.
#
# This file is distributed under the terms of the GNU General Public
# License (version 2.0 or later).
#
# The full license is available in LICENSE.GPL.txt, distributed with
# this software.
#
# SPDX-License-Identifier: GPL-2.0-or-later
# ------------------------------------------------------------------

# PyPylon is a tricky library to bundle. It encapsulates the pylon C++ SDK inside
# it with modified library references to make the module relocatable.
# PyInstaller is able to find those libraries and preserve the linkage for almost
# all of them. However - there is an additional linking step happening at runtime,
# when the library is creating the transport layer for the camera. This linking
# will fail with the library files modified by pyinstaller.
# As the module is already relocatable, we circumvent this issue by bundling
# pypylon as-is - for pyinstaller we treat the shared library files as just data.

import os
import PyInstaller.utils.hooks as hooks

# Patterns of dynamic library filenames that might be bundled with some installed Python packages.
# extended to cover the unsually named pylon-libusb
hooks.PY_DYLIB_PATTERNS = [
    '*.dll',
    '*.dylib',
    'lib*.so',
    'pylon*.so',
]

# Collect dynamic libs as data (to prevent pyinstaller from modifying them)
datas = hooks.collect_dynamic_libs('pypylon')

# Collect data files, looking for pypylon/pylonCXP/bin/ProducerCXP.cti, but other files may also be needed
datas += hooks.collect_data_files('pypylon')

# Exclude the C++-extensions from automatic search, add them manually as data files
# their dependencies were already handled with collect_dynamic_libs
excludedimports = ['pypylon._pylon', 'pypylon._genicam']
for filename, module in hooks.collect_all('pypylon._pylon')[0]:
    print(filename, module)
    if (os.path.basename(filename).startswith('_pylon.')
            or os.path.basename(filename).startswith('_genicam.')):
        datas += [(filename, module)]
```

sarperkilic, 2022-02-23:
> Hi @sarperkilic,
>
> use this hook-file from a folder, that you reference with --additional-hooks-dir Not beautiful but fixes the missing .so file
>
> ```python
> # [...]

Hello @thiesmoeller,

Thank you for your time and interest. I successfully converted it with the hook file you shared with me.

sarperkilic, 2022-02-24:
Hello again @thiesmoeller,

I have one more problem related this issue.

As I told you, I was able to convert python code by using the hook file you shared with me, in Linux-x86_64-Ubuntu-18.04 operating system.

But I faced the same problem in Linux-tegra-aarch64-with-Ubuntu-18.04 operating system.

I am deploying our application to nvidia jetson devices and I need a new hook file. Could you provide with me a suitable hook file for this problem? 

I am sharing the detailed logs below.

pyinstaller --onefile --additional-hooks-dir=. camera.py
943 INFO: PyInstaller: 4.2
943 INFO: Python: 3.6.9
945 INFO: Platform: Linux-4.9.253-tegra-aarch64-with-Ubuntu-18.04-bionic
946 INFO: wrote /home/User/Documents/exe_camera/camera.spec
951 INFO: UPX is not available.
956 INFO: Extending PYTHONPATH with paths
['/home/User/Documents/exe_camera',
 '/usr/lib/python3.6/dist-packages/cv2/python-3.6',
 '/home/User/Documents/exe_camera']
986 INFO: checking Analysis
986 INFO: Building Analysis because Analysis-00.toc is non existent
986 INFO: Initializing module dependency graph...
993 INFO: Caching module graph hooks...
1008 WARNING: Several hooks defined for module 'pypylon'. Please take care they do not conflict.
1031 INFO: Analyzing base_library.zip ...
14309 INFO: Caching module dependency graph...
14660 INFO: running Analysis Analysis-00.toc
14755 INFO: Analyzing /home/User/Documents/exe_camera/camera.py
17840 INFO: Processing pre-find module path hook distutils from '/usr/local/lib/python3.6/dist-packages/PyInstaller/hooks/pre_find_module_path/hook-distutils.py'.
17841 INFO: distutils: retargeting to non-venv dir '/usr/lib/python3.6'
24764 INFO: Processing pre-find module path hook site from '/usr/local/lib/python3.6/dist-packages/PyInstaller/hooks/pre_find_module_path/hook-site.py'.
24765 INFO: site: retargeting to fake-dir '/usr/local/lib/python3.6/dist-packages/PyInstaller/fake-modules'
35828 INFO: Processing module hooks...
35828 INFO: Loading module hook 'hook-pypylon.py' from '/home/User/Documents/exe_camera'...
/home/inovako/.local/lib/python3.6/site-packages/pkg_resources/__init__.py:119: PkgResourcesDeprecationWarning: 0.18ubuntu0.18.04.1 is an invalid version and will not be supported in a future release
  PkgResourcesDeprecationWarning,
35998 WARNING: Unable to copy metadata for pypylon._pylon: The 'pypylon._pylon' distribution was not found and is required by the application
36006 INFO: Determining a mapping of distributions to packages...
98514 WARNING: Unable to determine requirements for pypylon._pylon: The 'pypylon._pylon' distribution was not found and is required by the application
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylon_TL_gige-6.2.0.so](http://libpylon_tl_gige-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/libGenApi_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylonbase-6.2.0.so](http://libpylonbase-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylon_TL_gtc-6.2.0.so](http://libpylon_tl_gtc-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/libGCBase_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylon_TL_usb-6.2.0.so](http://libpylon_tl_usb-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/__pycache__/__init__.cpython-36.pyc pypylon/__pycache__
/usr/local/lib/python3.6/dist-packages/pypylon/libLog_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/__pycache__/genicam.cpython-36.pyc pypylon/__pycache__
/usr/local/lib/python3.6/dist-packages/pypylon/__init__.py pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylonutility-6.2.0.so](http://libpylonutility-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/genicam.py pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/pylon.py pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/_[pylon.cpython-36m-aarch64-linux-gnu.so](http://pylon.cpython-36m-aarch64-linux-gnu.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libgxapi-6.2.0.so](http://libgxapi-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/libNodeMapData_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[pylon-libusb-1.0.so](http://pylon-libusb-1.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/_[genicam.cpython-36m-aarch64-linux-gnu.so](http://genicam.cpython-36m-aarch64-linux-gnu.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/liblog4cpp_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/libMathParser_gcc_v3_1_Basler_pylon.so pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libpylon_TL_camemu-6.2.0.so](http://libpylon_tl_camemu-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/[libuxapi-6.2.0.so](http://libuxapi-6.2.0.so/) pypylon
/usr/local/lib/python3.6/dist-packages/pypylon/__pycache__/pylon.cpython-36.pyc pypylon/__pycache__
/usr/local/lib/python3.6/dist-packages/pypylon/libXmlParser_gcc_v3_1_Basler_pylon.so pypylon
98532 INFO: Loading module hook 'hook-lib2to3.py' from '/usr/local/lib/python3.6/dist-packages/PyInstaller/hooks'...
98592 INFO: Loading module hook '[hook-numpy.core.py](http://hook-numpy.core.py/)' from '/usr/local/lib/python3.6/dist-packages/PyInstaller/hooks'...
[... cut ...]
```

and the error is:

./dist/camera

```
globbing failed
Traceback (most recent call last):
  File "camera.py", line 5, in <module>
    cam = py.InstantCamera(tlf.CreateFirstDevice())
  File "pypylon/pylon.py", line 1719, in CreateFirstDevice
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
[28857] Failed to execute script camera
```

I am looking forward to your reply.
Thanks

sarperkilic, 2022-02-25:
> complicated

:) Jetson is target device to execute our program. 

I run the code without --onefile, below I share the files in the dist director
[image]
[image]

thiesmoeller, 2022-02-25:
on our jetson nano system this works ... ( with above modified hook file )
attached is a onefile aarch64/ubuntu 18.04 version, that works on our side.

[test_enum.zip](https://github.com/basler/pypylon/files/8141134/test_enum.zip)

sarperkilic, 2022-02-25:
> on our jetson nano system this works ... ( with above modified hook file ) attached is a onefile aarch64/ubuntu 18.04 version, that works on our side.
> 
> [test_enum.zip](https://github.com/basler/pypylon/files/8141134/test_enum.zip)

Hello,

this onefile is worked on me. 
this is the output:
/test_enum
acA1920-150um

But, when I try to convert my python script with above modified hook file, I got same error:
```
globbing failed
Traceback (most recent call last):
  File "camera.py", line 5, in <module>
    cam = py.InstantCamera(tlf.CreateFirstDevice())
  File "pypylon/pylon.py", line 1719, in CreateFirstDevice
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
[11741] Failed to execute script camera
```

What am i doing different?
Are you sure, you used the above hook file to create onefile test_enum?

thiesmoeller, 2022-02-25:
a) what is this message "globbing failed" ...
b) Use a virtual or containerized environment ... limit the amount of dependencies into your project
 I created this in an arm64 ubuntu-core 18.04 docker environment to be sure that  everything is clean

sarperkilic, 2022-02-28:
> a) what is this message "globbing failed" ... b) Use a virtual or containerized environment ... limit the amount of dependencies into your project I created this in an arm64 ubuntu-core 18.04 docker environment to be sure that everything is clean

Hello again @thiesmoeller,

I have created a new virtual environment to run pyinstaller command.

I have installed only pyinstaller, numpy and opencv libraries through this virtual environment. and then I run this command:
`pyinstaller --additional-hooks-dir=. basler_image.py`

This is the dist folder:
[image]

And when I try to exec the output file, I got this error:

[image]

This is my basler_image.py file:
```
import pypylon.pylon as py
tlf = py.TlFactory.GetInstance()
cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()
res = cam.GrabOne(1000)
img = res.Array
print("captured image: pixel_0,0 ", img[0,0])
cam.Close()
```

This is my hook file:

```
import os
import PyInstaller.utils.hooks as hooks

# Patterns of dynamic library filenames that might be bundled with some installed Python packages.
# extended to cover the unsually named pylon-libusb
hooks.PY_DYLIB_PATTERNS = [
    '*.dll',
    '*.dylib',
    'lib*.so',
    'pylon*.so',
]

# Collect dynamic libs as data (to prevent pyinstaller from modifying them)
datas = hooks.collect_dynamic_libs('pypylon')

# Collect data files, looking for pypylon/pylonCXP/bin/ProducerCXP.cti, but other files may also be needed
datas += hooks.collect_data_files('pypylon')

# Exclude the C++-extensions from automatic search, add them manually as data files
# their dependencies were already handled with collect_dynamic_libs
excludedimports = ['pypylon._pylon', 'pypylon._genicam']
for filename, module in hooks.collect_all('pypylon._pylon')[0]:
    print(filename, module)
    if (os.path.basename(filename).startswith('_pylon.')
            or os.path.basename(filename).startswith('_genicam.')):
        datas += [(filename, module)]

```

I am able to run basler_image.py script successfully from terminal with command python3 basler_image.py.

What do you suggest me to solve this problem?

And I am also sharing the pyinstaller logs below:

```
158 INFO: PyInstaller: 4.2
159 INFO: Python: 3.6.9
161 INFO: Platform: Linux-4.9.253-tegra-aarch64-with-Ubuntu-18.04-bionic
[... 83 lines cut ...]
40660 INFO: Looking for dynamic libraries
42202 INFO: Looking for eggs
42203 INFO: Python library not in binary dependencies. Doing additional searching...
42291 INFO: Using Python library /usr/lib/aarch64-linux-gnu/libpython3.6m.so.1.0
42327 INFO: Warnings written to /home/User/Documents/venv/build/basler_image/warn-basler_image.txt
42594 INFO: Graph cross-reference written to /home/User/Documents/venv/build/basler_image/xref-basler_image.html
42665 INFO: checking PYZ
42665 INFO: Building PYZ because PYZ-00.toc is non existent
42665 INFO: Building PYZ (ZlibArchive) /home/User/Documents/venv/build/basler_image/PYZ-00.pyz
46020 INFO: Building PYZ (ZlibArchive) /home/User/Documents/venv/build/basler_image/PYZ-00.pyz completed successfully.
46072 INFO: checking PKG
46073 INFO: Building PKG because PKG-00.toc is non existent
46073 INFO: Building PKG (CArchive) PKG-00.pkg
46194 INFO: Building PKG (CArchive) PKG-00.pkg completed successfully.
46197 INFO: Bootloader /home/User/Documents/venv/exec_env/lib/python3.6/site-packages/PyInstaller/bootloader/Linux-64bit-aarch/run
46197 INFO: checking EXE
46197 INFO: Building EXE because EXE-00.toc is non existent
46198 INFO: Building EXE from EXE-00.toc
46198 INFO: Appending archive to ELF section in EXE /home/User/Documents/venv/build/basler_image/basler_image
46227 INFO: Building EXE from EXE-00.toc completed successfully.
46231 INFO: checking COLLECT
46232 INFO: Building COLLECT because COLLECT-00.toc is non existent
46232 INFO: Building COLLECT COLLECT-00.toc
46694 INFO: Building COLLECT COLLECT-00.toc completed successfully.
```

thiesmoeller, 2022-03-01:
From where do you get your pyinstaller?
You use **pyinstall 4.2**
When we install, as described in the examples, via pip.  we get **version 4.9**
This is the only delta I can spot between yours and our logs.
Is it a mix of ubuntu system packages and pip installed packages?

so for the simple test only put this into your venv 
```
python3 -m pip install --upgrade pip
pip3 install pypylon pyinstaller
```

sarperkilic, 2022-03-17:
> @sarperkilic I hope this question can be now closed?

Hello,

No, the problem is not solved.

I am still having the same error.

SMA2016a, 2022-03-17:
since it is working on your AGX board, please upload your SD-Card backup image. I will check it. Otherwise, I do not see a better way to help you on this issue.
On my Jetson nano und in venv it is working

IgnacioBarroso, 2022-06-10:
I was have the same problem, you need to try with THIS hook, and this import **import pypylon.pylon as pylon**. Its worked in my case. If still not working, the problem is a diff between your import syntax and the way to write the filename of the hook. Code around that, its not a compatibility problem, only syntax.
`# ------------------------------------------------------------------
# Copyright (c) 2020 PyInstaller Development Team.
#
# This file is distributed under the terms of the GNU General Public
# License (version 2.0 or later).
#
# The full license is available in LICENSE.GPL.txt, distributed with
# this software.
#
# SPDX-License-Identifier: GPL-2.0-or-later
# ------------------------------------------------------------------

# PyPylon is a tricky library to bundle. It encapsulates the pylon C++ SDK inside
# it with modified library references to make the module relocatable.
# PyInstaller is able to find those libraries and preserve the linkage for almost
# all of them. However - there is an additional linking step happening at runtime,
# when the library is creating the transport layer for the camera. This linking
# will fail with the library files modified by pyinstaller.
# As the module is already relocatable, we circumvent this issue by bundling
# pypylon as-is - for pyinstaller we treat the shared library files as just data.

import os
import PyInstaller.utils.hooks as hooks

# Patterns of dynamic library filenames that might be bundled with some installed Python packages.
# extended to cover the unsually named pylon-libusb
hooks.PY_DYLIB_PATTERNS = [
    '*.dll',
    '*.dylib',
    'lib*.so',
    'pylon*.so',
]

# Collect dynamic libs as data (to prevent pyinstaller from modifying them)
datas = hooks.collect_dynamic_libs('pypylon')

# Collect data files, looking for pypylon/pylonCXP/bin/ProducerCXP.cti, but other files may also be needed
datas += hooks.collect_data_files('pypylon')

# Exclude the C++-extensions from automatic search, add them manually as data files
# their dependencies were already handled with collect_dynamic_libs
excludedimports = ['pypylon._pylon', 'pypylon._genicam']
for filename, module in hooks.collect_all('pypylon')[0]:
    if (os.path.basename(filename).startswith('_pylon.')
            or os.path.basename(filename).startswith('_genicam.')):
        datas += [(filename, module)]`

IgnacioBarroso, 2022-06-10:
idk how to show markdown like a raw. Whatever, is logic, copy and paste and be happy(?

k-sokolov, 2023-01-17:
had the same issue as the topicstarter when using python3.8 on x86 ubuntu 18.04.
applying this hook
https://github.com/basler/pypylon/issues/36#issuecomment-771205790
solved the issue. 

> Below is a simplified test script and a Dockerfile to test on 18.04 in clean environment. This works.
>
> Can you do the same test on your system, to just use the CAMEMU camera, as this will prove that [...]

@thiesmoeller this test was working fine but when I modified the dockerfile to install python3.8 instead of 3.6, the issue was reproduced.

thiesmoeller, 2023-01-18:
@k-sokolov did you install a different python version or simply switch to a Ubuntu 20.04 base?

k-sokolov, 2023-01-18:
> @k-sokolov did you install a different python version or simply switch to a Ubuntu 20.04 base?

installed python3.8, base was still 18.04

k-sokolov, 2023-01-18:
@thiesmoeller just checked, changing the base to 20.04 also results in the error
```
Step 7/9 : RUN PYLON_CAMEMU=1 dist/issue_464
 ---> Running in 7695b0e054d1
globbing failed
Traceback (most recent call last):
  File "issue_464.py", line 5, in <module>
  File "pypylon/pylon.py", line 1601, in CreateFirstDevice
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
[8] Failed to execute script 'issue_464' due to unhandled exception!
```

IgnacioBarroso, 2023-04-03:
The cameras fail to capture randomly and with a proportion of 40% of images with errors. It happens to me with all the linear cameras that we have bought for different projects. What should i do for its correct configuration if that is the problem. Thanks!​

Ignacio Barroso
Programador, Ingeniería Elementos Combustibles.
CONUAR S.A.

thiesmoeller, 2023-04-03:
@IgnacioBarroso,  if your issue is not regarding pyinstaller please open a new issue to properly track your questions

BryanDiLaura, 2025-06-03:
For anyone else who runs into this problem, I was able to get a workaround using the recommendations in [this thread](https://github.com/basler/pypylon/issues/36#issuecomment-771205790), which suggests a custom hook to relocate a few of the shared objects.

From my naive understanding it seems like there's something off about the way that pypylon's shared objects are collected and how those are packaged in pyinstaller. As far as I can tell the only real difference is in directory structure/organization. I don't know if there was some change in they way pylon organizes things or how pyinstaller is doing things (maybe they got out of sync at some point?), but someone else who is more familiar with this stuff may know better. 

For future reference, I'm using:

- Ubuntu 24.04
- python 3.13.3
- pyinstaller==6.12.0 (uses the `pyinstaller-hooks-contrib` for the pypylon hook)
- Pylon 8.1.0

And I was observing effectively a silent failure where running the following script enumerated a couple devices using my system version of everything, but the pyinstaller version returned an empty enumeration with "globbing failed" printed to stdout.

```python
from pypylon import pylon
tlf = pylon.TlFactory.GetInstance()
tlf.EnumerateDevices()
```

Hopefully this helps!