basler/pypylon #465: segmentation fault while initializing multiple cameras and grabbing the frames.

karankhatri13, 2022-02-03:
I'm currently working with multiple cameras for my computer vision project. I had created a script for initiating all the camera's at once and use specific camera that is called and grab the image from that specific camera.

```
from pypylon import pylon

class PylonCamera:
    def __init__(self):
        self.tl_factory = pylon.TlFactory.GetInstance()
        self.camera_ip = []
        self.camera_list = {}

        devices = self.tl_factory.EnumerateDevices()
        for camera_info in self.tl_factory.EnumerateDevices():
            if camera_info.GetDeviceClass() == "BaslerGigE":
                self.camera_ip.append(camera_info.GetIpAddress().split('.')[-1])

        cameras = pylon.InstantCameraArray(len(devices))

        for i, cam in enumerate(cameras):
            cam.Attach(self.tl_factory.CreateDevice(devices[i]))
            cam.Open()
            pylon.FeaturePersistence.Load(
                self.camera_ip[i] + ".pfs", cam.GetNodeMap(), True
            )
            cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

            self.camera_list[self.camera_ip[i]] = cam

    def get_image(self, camera_ip):
        # Grabing Continusely (video) with minimal delay
        grab = self.camera_list[camera_ip].RetrieveResult(
            2000, pylon.TimeoutHandling_Return
        )
        img = None

        if grab.GrabSucceeded():
            img = grab.GetArray()

            grab.Release()

        return img
```

In short, I am taking all the connected devices, initiated all the camera and created a dictionary with data like key is ip of camera and value is the camera object. 

the issue is when I call the get_image function there is segmentation fault core dumped error but code works when I write get image code in the __init__ function.

thiesmoeller, 2022-02-03:
Please provide some debug info from your segfault.
Where does this happen? What platform?

Notes:

* `grab.Release()` hast to be called also in grab failed path

* Your are mixing this internal InstantCameraArray ... with your own camera dict ...
   Recommend to use just one style like in:

```python
import os
import pypylon.pylon as py
import numpy as np

os.environ["PYLON_CAMEMU"] = "10"

class PylonCameras(object):
    def __init__(self):
        self._tl_factory = py.TlFactory.GetInstance()
        self._camera_list = {}

        devices = self._tl_factory.EnumerateDevices()
        for di in self._tl_factory.EnumerateDevices():
            cam = py.InstantCamera(self._tl_factory.CreateDevice(di))
            cam.Open()
            # py.FeaturePersistence.Load( ....
            cam.StartGrabbing(py.GrabStrategy_LatestImageOnly)
            print("add ", di.GetSerialNumber())
            self._camera_list[di.GetSerialNumber()] = cam

    def get_image(self, serial_num: str):
        # Grabbing Continuously (video) with minimal delay
        img = None
        try:
            grab = self._camera_list[serial_num].RetrieveResult(
                2000, py.TimeoutHandling_ThrowException
            )
            if grab.GrabSucceeded():
                img = grab.GetArray()

            grab.Release()
        except py.TimeoutException as e:
            print("Timeout")

        return img

cams = PylonCameras()

for i in range(10):
    print(cams.get_image("0815-0006"))

```

karankhatri13, 2022-02-03:
@thiesmoeller  I had tried debugging the error log but unables to find get it. I am getting the segmentation fault while grabbing the frame `grab = self.camera_list[camera_ip].RetrieveResult(2000, pylon.TimeoutHandling_Return)`. 

Thanks for the quick reply. I will try the solution you mentioned.

karankhatri13, 2022-02-03:
@thiesmoeller i had tried your code and exactly what I want and works perfectly fine without loading feature file. But the issue arises when I load feature file.

> Traceback (most recent call last):
  File "multi_github.py", line 39, in <module>
    cams = PylonCameras()
  File "multi_github.py", line 16, in __init__
    py.FeaturePersistence.Load("feature.pfs", cam.GetNodeMap(), True)
  File "Python\Python38\lib\site-packages\pypylon\pylon.py", line 7071, in Load
    return _pylon.FeaturePersistence_Load(FileName, pNodeMap, validate)
_genicam.RuntimeException: Error while writing node: 'GainAuto' not found
Error while writing node: 'GainSelector' not found
Error while writing node: 'GainSelector' not found
Error while writing node: 'BlackLevelSelector' not found
Error while writing node: 'BlackLevelRaw' not : RuntimeException thrown (file 'featurepersistence.cpp', line 64)

thiesmoeller, 2022-02-03:
my sample code is using the emulated cameras for a quick test without real cameras
your pfs file is special to your camera models..

... this should work on your side .... adding a filter for GigE devices to the enumeration process

```python
import pypylon.pylon as py

class PylonCameras(object):
    def __init__(self):
        self._tl_factory = py.TlFactory.GetInstance()
        self._camera_list = {}

        # setup device filter to only show GigE devices
        di_filter = py.DeviceInfo()
        di_filter.SetDeviceClass("BaslerGigE")

        devices = self._tl_factory.EnumerateDevices([di_filter])
        for di in devices:
            cam = py.InstantCamera(self._tl_factory.CreateDevice(di))
            cam.Open()
            # py.FeaturePersistence.Load( ....
            cam.StartGrabbing(py.GrabStrategy_LatestImageOnly)
            print("add ", di.GetSerialNumber(), di.GetDeviceClass())
            self._camera_list[di.GetSerialNumber()] = cam

    def get_image(self, serial_num: str):
        # Grabbing Continuously (video) with minimal delay
        img = None
        try:
            grab = self._camera_list[serial_num].RetrieveResult(
                2000, py.TimeoutHandling_ThrowException
            )
            if grab.GrabSucceeded():
                img = grab.GetArray()

            grab.Release()
        except py.TimeoutException as e:
            print("Timeout")

        return img

cams = PylonCameras()

for i in range(10):
    print(cams.get_image("23423423423"))

```

closed 2022-02-03 by karankhatri13