basler/pypylon #466: Most efficient way to grab and save images to disk at even intervals

milancurcic, 2022-02-03:
Hi, thank you for this great library. I'm developing an image recorder using

* Camera: Basler aca2440-35um (5MP, 35 fps)
* Board: Jetson Nano
* Disk: Samsung T5 USB3 SSD
* OS: Ubuntu 18.04
* Pylon version: 6.2.0.21487-deb0_arm64

My target is to acquire 5MP at 20 fps.

## Writing to TIFF using `pylon.PylonImage().Save`

To start, I adapted the code from the samples, and I use it to grab from the camera and write to TIFF. I just let the script grab and write as fast as it can, no scheduling:

```python
import argparse
from pypylon import pylon
import time

parser = argparse.ArgumentParser()
parser.add_argument('width', type=int, help='Image width (pixels)')
parser.add_argument('height', type=int, help='Image height (pixels)')
parser.add_argument('n', type=int, help='How many images to take')
args = parser.parse_args()

height, width, n, = args.height, args.width, args.n
image_size = height * width

output_path = '/mnt/spray-imager-storage/images/'

img = pylon.PylonImage()

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()
cam.Height.SetValue(height)
cam.Width.SetValue(width)
cam.StartGrabbing()

for m in range(n):
    with cam.RetrieveResult(2000) as res:
        t1 = time.time()

        img.AttachGrabResultBuffer(res)
        img.Save(pylon.ImageFileFormat_Tiff, output_path + 'test.tif')

        t2 = time.time()
        print('fps: %.2f' % (1 / (t2 - t1)), \
              'bandwidth: %.2f MB/s' % (image_size / 1024**2 / (t2 - t1)))

cam.StopGrabbing()
cam.Close()
```

Output:

```
$ PYLON_CAMEMU=0 python3 grab_img_write_ssd.py 2448 2048 10
fps: 6.30 bandwidth: 30.12 MB/s
fps: 6.75 bandwidth: 32.26 MB/s
fps: 6.72 bandwidth: 32.13 MB/s
fps: 6.79 bandwidth: 32.47 MB/s
fps: 6.83 bandwidth: 32.65 MB/s
fps: 6.77 bandwidth: 32.36 MB/s
fps: 6.72 bandwidth: 32.13 MB/s
fps: 6.82 bandwidth: 32.63 MB/s
fps: 6.89 bandwidth: 32.92 MB/s
fps: 6.87 bandwidth: 32.85 MB/s
```

Note that in this script, I always write to `test.tif` on the SSD. If I change the filename so that each files gets its own file, e.g.

```
img.Save(pylon.ImageFileFormat_Tiff, output_path + 'test_%i.tif' % m)
```

the speed degrades 3-fold:

```
$ PYLON_CAMEMU=0 python3 grab_img_write_ssd.py 2448 2048 10
fps: 2.31 bandwidth: 11.03 MB/s
fps: 2.37 bandwidth: 11.35 MB/s
fps: 2.38 bandwidth: 11.36 MB/s
fps: 2.40 bandwidth: 11.48 MB/s
fps: 2.38 bandwidth: 11.38 MB/s
fps: 2.37 bandwidth: 11.35 MB/s
fps: 2.44 bandwidth: 11.65 MB/s
fps: 2.45 bandwidth: 11.70 MB/s
fps: 2.44 bandwidth: 11.68 MB/s
fps: 2.42 bandwidth: 11.59 MB/s
```

Why is this? I can't explain it, other than either Python runtime, or the OS, does something under the hood to make things faster when writing to the same file.

## Writing to TIFF using `PIL.Image`

Then I used an alternative writer:

```python
import argparse
from datetime import datetime
from pypylon import pylon
from io import BytesIO
from PIL import Image
import time

parser = argparse.ArgumentParser()
parser.add_argument('width', type=int, help='Image width (pixels)')
parser.add_argument('height', type=int, help='Image height (pixels)')
parser.add_argument('n', type=int, help='How many images to take')
args = parser.parse_args()

height, width, n, = args.height, args.width, args.n
image_size = height * width

output_path = '/mnt/spray-imager-storage/images/'

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()
cam.Height.SetValue(height)
cam.Width.SetValue(width)
cam.StartGrabbing()

for m in range(n):
    with cam.RetrieveResult(2000) as res:
        t1 = time.time()

        buf = BytesIO()
        Image.fromarray(res.Array).save(buf, 'tiff')
        now = datetime.utcnow()
        filename = 'img_' + now.strftime('%Y%m%d_%H%M%S.%f.tif')
        open(output_path + filename, 'wb').write(buf.getbuffer())

        t2 = time.time()
        print('fps: %.2f' % (1 / (t2 - t1)), \
              'bandwidth: %.2f MB/s' % (image_size / 1024**2 / (t2 - t1)))

cam.StopGrabbing()
cam.Close()
```

and the output:

```
$ PYLON_CAMEMU=0 python3 grab_plain_tiff_write_ssd.py 2448 2048 10
fps: 13.53 bandwidth: 64.70 MB/s
fps: 28.25 bandwidth: 135.08 MB/s
fps: 26.04 bandwidth: 124.53 MB/s
fps: 26.47 bandwidth: 126.55 MB/s
fps: 25.85 bandwidth: 123.60 MB/s
fps: 25.46 bandwidth: 121.74 MB/s
fps: 27.08 bandwidth: 129.46 MB/s
fps: 26.55 bandwidth: 126.94 MB/s
fps: 26.69 bandwidth: 127.62 MB/s
fps: 26.73 bandwidth: 127.79 MB/s
```

Now this is much better and meets my spec, except that during longer acquisition times, fps dips to ~13 for a short while and then goes back to ~26 fps. I assume some OS stuff or filesystem maintenance getting in the way. Also note that in this program I already have filenames with their own timestamps and it doesn't affect the write speed.

If I measure write speed to my SSD with `dd`, I get ~200 MB/s for 5MB blocksize. However, it's possible that the USB ports on Jetson Nano share the bus, so maybe the camera and the SSD share the bandwidth (just a guess, I don't know).

## Writing to plain binary instead of TIFF

I also tried to dump the plain binary data to file instead of encoding to TIFF, as I can afford converting binary to TIFF after the fact. This shaves 8ms off the total elapsed time to process a frame.

## Questions

* What's your recommended way to save images to TIFF?
* What's the appropriate grabbing strategy for my use case (grab an image every 50 ms)?
* Does PyPylon provide any software trigger to grab, say, every 50 ms? So far I used [apscheduler](https://apscheduler.readthedocs.io/en/3.x/) which is okay for every 1 s, but at higher frequency it misses the mark.

Thank you!

thiesmoeller, 2022-02-04:
a) 
you should optimize the USB performance according to the readme, when you install full pylon SDK
`camera.StreamGrabber.MaxTransferSize = 4194304`

... from the full readme of pylon SDK:

`
* Increasing Packet Size
  For faster USB transfers, you should increase the packet size. If you're
  working with the pylon Viewer, you can do this by changing the
  "Stream Parameters" -> "Maximum Transfer Size" value from inside the
  pylon Viewer. If you're writing your own application, use the camera API
  to set the "Maximum Transfer Size".
  After increasing the package size, you will likely run out of kernel space
  and see corresponding error messages in the console. The default value set
  by the kernel is 16 MB. To change the value (in this example to 1000 MB), you can
  execute as root:
    echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb
  This would assign a maximum of 1000 MB to the USB stack.
`

b)
The camera can generate very precise framerate. There is a feature you have to configure

https://docs.baslerweb.com/acquisition-frame-rate?filter=Camera:acA2440-35um

```python
# Set the upper limit of the camera's frame rate to 20 fps
camera.AcquisitionFrameRateEnable.SetValue(True)
camera.AcquisitionFrameRate.SetValue(20.0)
```

c) 
grab strategy is fine... camera enforces 20fps

milancurcic, 2022-02-04:
Thanks a lot @thiesmoeller, you rock.

a) seems like it helped somewhat, though not quite remove the issue completely. Every few seconds I see a dip in FPS (by about a factor of two) for few-to-several frames, then goes back to normal. I will try to diagnose in more depth what happens when that dip happens.

b) did the job, thanks! Though it looks like I may need to switch to a hardware trigger so I can sync with another device.

I will keep this issue open while I track down a) and will close once resolved.

thiesmoeller, 2022-02-04:
What you measure is not fps but inverse of processing time of your save routine.

The camera will exactly deliver the configured framerate.

milancurcic, 2022-02-05:
You're correct, all clear, thanks again.

closed 2022-02-05 by milancurcic