basler/pypylon #467: Using Test Images with emulated camera

barliya, 2022-02-05:
Hi Guys,

I am trying to use the test images feature of the emulated camera to test my procedures.
I am using python 3.9.9, and my code recognizing I am using an emulated camera rather than a real one.

However, I am not able to generate the test images (although, it was working for me once).
When writing this:
`camera.TestImageSelector.SetValue("TestImageSelector_Testimage2")` 
(or any other image for that matter), I get the the following error:

File "/usr/local/lib/python3.9/site-packages/pypylon/genicam.py", line 2316, in SetValue
    return _genicam.IEnumeration_SetValue(self, entry)
_genicam.InvalidArgumentException: Feature 'TestImageSelector' : cannot convert value 'TestImageSelector_Testimage2', the value is invalid. : InvalidArgumentException thrown in node 'TestImageSelector' while calling 'TestImageSelector.FromString()' (file 'Enumeration.cpp', line 134)

I would appreciate any help on that matter.
Thanks

thiesmoeller, 2022-02-05:
It's just "Testimage2"

SMA2016a, 2022-05-17:
@barliya please close this question if you solved this.

closed 2022-05-18 by barliya