basler/pypylon #468: _genicam.OutOfRangeException: Value = 1000 must be equal or smaller than Max = 8.

zoldaten, 2022-02-08:
i work with acA4112-20um usb camera.
when i use options:
```
camera.Width.SetValue(800) # 4096 px x 3000 px
camera.Height.SetValue(3000)
```
everything works as expected as camera has 4096x3000 resolution.
But when i add offsets:
```
camera.OffsetX.SetValue(1800);
camera.OffsetY.SetValue(1000);
````
I crashes with error:
```
Traceback (most recent call last):
  File "/home/pi/Desktop/test-zbar/test_basler3.py", line 31, in <module>
    camera.OffsetY.SetValue(1000);
  File "/home/pi/.local/lib/python3.7/site-packages/pypylon/genicam.py", line 2159, in SetValue
    return _genicam.IInteger_SetValue(self, Value, Verify)
_genicam.OutOfRangeException: Value = 1000 must be equal or smaller than Max = 8. : OutOfRangeException thrown in node 'OffsetY' while calling 'OffsetY.SetValue()' (file 'IntegerT.h', line 79)
```
To work with offset i should use maximum 2008 px:
```
camera.Width.SetValue(800) # 4096 px x 3000 px
camera.Height.SetValue(2008)
```
Is it bag or i missed something ?
And how to return defaults of the camera because sometimes it remembers the options and can`t switch back to 800x3000 resolution?

thiesmoeller, 2022-02-08:
to have your camera in the default state every time load the factory first:

```python
....
cam.Open()
cam.UserSetSelector = "Default"
cam.UserSetLoad.Execute()
....
```

thiesmoeller, 2022-02-10:
Can you post your full configuration sequence code including the reset to defaults I wrote above to see the error.

It seems that your seting the Height never gets active as 3008 is the real height 3000 wake-up height.

thiesmoeller, 2022-02-10:
You set a height of 3000.

OffsetY is the top row of the RegionOfInterest

As the sensor has only 3008 rows, you can't do this...as your requested ROI would overlap the sensor. ( Last row of image would be 3000 + 1000.. )

You can use the pylonviewer do visually "play" with the settings

zoldaten, 2022-02-10:
got it, thanks.