basler/pypylon #469: GigE camera can't sync with light trigger

maxwxzheng, 2022-02-08:
I'm developing an imaging product where we need to sync the light and the camera. I'm working with an acA3800-10gm GigE camera, and an acA2440-75uc USB3 camera.
When we trigger our light, we send a trigger signal to the camera immediately. Then we keep the light on for the duration that is the same as the exposure set on the camera. This works well with the USB3 camera. We are able to get clear images. However, with the GigE camera, we always get a dark image. I've tried exposure ranging from 35us to 3500us on the GigE camera. I suspect that the camera is capturing images either too early, or too late.

Here is the code we initialize the cameras with.
```
camera.Open()
camera.AcquisitionStatusSelector = "FrameTriggerWait"
camera.TriggerSelector.SetValue("FrameStart")
camera.TriggerMode.SetValue("On")
camera.TriggerSource.SetValue("Line1")
camera.TriggerActivation.SetValue('RisingEdge')
```

The debouncer on both cameras are 0.
The trigger on both cameras are 0.

Anything else I can try to debug this?

thiesmoeller, 2022-02-09:
You are aware that you mix a global and a rolling shutter camera?

If this is what you want...
you would have to 

A) check the different exposure start delays of the cameras on  docs.baslerweb.com 
They can be very different between rolling and global shutter cameras!
(ESD Time from trigger to start of exposure )
To align the times you can use the triggerdelay feature on the faster camera.
B) calculate the flash window position on the rolling shutter camera ( start delay + readout time) and set the exposure time > readouttime
Flash window also described on docs.baslerweb.com

The global shutter camera, to be in sync with the rolling, has to be delayed until the rolling shutter flashwindow opens!

maxwxzheng, 2022-02-09:
Thanks @thiesmoeller!
The Exposure Start Delay makes sense to me. For acA3800-10gm, it is 2900 us. I understand that after the hard trigger, the camera will start the exposure after 2900 us.
I don't fully understand the Flash Window concept for rolling shutter camera. Should we only turn on the light during the flash window (when all rows are open for exposure)?

thiesmoeller, 2022-02-09:
Yes. Rolling shutter and flash only works in flash window.

Only exception is complete darkness. Then you can use global reset release mode.

Here the exposure starts like global shutter but readout is rolling.
So ambient light would be evil as the image would get brighter and brighter in every row...

Concepts are documented in docs.baslerweb.com

maxwxzheng, 2022-02-09:
So if I want to build a lighting system that works with a rolling shutter camera, would this be how it works?
1. At t0, we trigger the camera.
2. At (t0 + Exposure Start Delay), the first row on the sensor starts exposure.
3. At (t0 + Exposure Start Delay + tRow * (Image ROI Height - 1)), Flash Window starts, and this is the time when our light should turn on.
4. At (t0 + Exposure Start Delay + exposure), Flash Window ends, and this is the time when our light should turn off.
Is this correct?

maxwxzheng, 2022-02-09:
I also did another experiment.
I changed the acA3800-10gm to use GlobalRestRelease mode.
Then I sent a trigger to the camera with exposure time 3,500us. I turned on the light at the same time and leave it on for 60,000 us. According to my understanding of GlobalRestRelease mode, the Flash Window should start at 2,900 us. So 60,000 us of light-on should cover the whole Flash Window. For some reason, we are still getting black images. Anything I did wrong/misunderstood here?

thiesmoeller, 2022-02-09:
If the exposure time on a rolling shutter camera is greater than the readouttime (your formulas are correct for this and  you can get this from the feature ReadoutTime) then the flashwindow "opens"

If ReadoutTime is e.g. 20ms and you set exposure time to 30ms you get a flash window of 10ms

Typically on rolling shutter you let the camera control the light.

This is why we offer on the digital output the Flashwindow signal, which gets active exactly in the window