basler/pypylon #470: Synchronously acquire images with multiple cameras

sarperkilic, 2022-02-09:
Hello eveyone,

I am using pypylon to take images from my application.

I will use 2 cameras and I need to grab images synchronously.

When I have a look to grabmultiplecameras sample, It says thay hardware trigger setup can be used to cause all cameras to grab images synchronously. According to their default configuration, the cameras are set up for free-running continuous acquisition.

Can I do this from pypylon? Or I have to do this on pylon viewer?

I am little bit lost about grabbing multiple images from multiple cameras synchronously from my pypylon script.

Detailed suggestions will be extremely useful for me.

Thanks

thiesmoeller, 2022-02-09:
Hi @sarperkilic, 

Please check our companion repository https://github.com/basler/pypylon-samples , where you'll find samples/notebooks both for multi camera and hardware trigger.

As an overall statement to _high precision_ syncing of cameras:

U3V: hardware
GEV: hardware, action command, ieee1588 synchronized freerun

sarperkilic, 2022-02-17:
> Hi @sarperkilic,
>
> Please check our companion repository https://github.com/basler/pypylon-samples , where you'll find samples/notebooks both for multi camera and hardware trigger.
>
> As an overall [...]

Hello again @thiesmoeller,

I need your help about this topic.

We want to take images by triggering two Basler color cameras synchronously.

We will do the triggering using pypylon.

When we connect two cameras to an industrial computer with a USB 3.0 port without the need for any additional cards (without an extra frame grabber GigE card), will we be able to run these cameras synchronously with a software trigger?

Are there any disadvantages of triggering with software trigger?

Are there any disadvantages to using USB 3.0 instead of GigE?

Are there any disadvantages to using the computer's USB 3.0 ports instead of the USB 3.0 Interface Card?

In other words, will it be technically possible for us to provide simultaneous software triggers, take pictures of two cameras at the same time, and transmit the pictures they took to the computer simultaneously?

Thanks in advance for your answers, if you need additional information, please let me know.

yctsai-metgauge, 2022-02-18:
Hi
I have the same problem about this question.
I am using two Basler aCA3800-10gm and would like to acquire the image synchronously of these two (or even more) cameras.
According to  IEEE1588, the synchronize accuracy is less than a microsecond.
However, my result shows that in both software and hardware trigger, the synchronization sometimes fail.
My experiment set up:

**trigger frequency** : 10fps

code:
```=python
from pypylon import pylon
import numpy as np
import matplotlib.pyplot as plt
import cv2
import os
import time

maxCamerasToUse = 2

tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices()
for device in devices:
    if device.GetDeviceClass() == 'BaslerGigE':
        print(f"using {device.GetFullName()} @ {device.GetIpAddress()}")
cam_array = pylon.InstantCameraArray(min(len(devices), maxCamerasToUse))

# action_key = 0x29
# group_key = 0x112233
# # group_mask = pylon.AllGroupMask
# group_mask = 0xffffffff
# # broadcastAddress = "192.168.50.255"

def using_cam_dic(cam_dic,idx,cam):
    cam_dic[idx]={
        "serialNum":cam.DeviceInfo.GetSerialNumber(),
        "MAC": cam.DeviceInfo.GetMacAddress(),
        "photo_id":0
    }

cam_dic={}
for idx, cam in enumerate(cam_array):
    cam.Attach(tl_factory.CreateDevice(devices[idx]))
    # cam.RegisterConfiguration(
    # pylon.ActionTriggerConfiguration(action_key, group_key, group_mask),
    # pylon.RegistrationMode_Append,
    # pylon.Cleanup_Delete
    # )
    cam.SetCameraContext(idx)
    print(f"target_cam-id:{idx} {cam.DeviceInfo.GetSerialNumber()}")
    using_cam_dic(cam_dic,idx,cam)

    
    cam.Open()

    cam.PixelFormat = "Mono8"
    cam.AcquisitionFrameRateEnable.SetValue(True)
    cam.AcquisitionFrameRateAbs.SetValue(50)

    cam.Width=512
    cam.Height=512
    cam.TriggerSelector.SetValue('FrameStart')
    cam.TriggerMode.SetValue('On')
    # cam.TriggerSource.SetValue("Action1")
    cam.TriggerSource.SetValue("Line1")
    print("ActionDeviceKey", hex(cam.ActionDeviceKey.GetValue()))
    print("ActionGroupKey", hex(cam.ActionGroupKey.GetValue()))
    print("ActionGroupMask", hex(cam.ActionGroupMask.GetValue()))
    print("TriggerSource", cam.TriggerSource.GetValue())
    print("TriggerMode", cam.TriggerMode.GetValue())
    print("TriggerActivation", cam.TriggerActivation.GetValue())
    print("AcquisitionMode", cam.AcquisitionMode.GetValue())
    print('\n')

img = pylon.PylonImage()
cam_index=[0,0]

cam_array.StartGrabbing()

print(f"Start Grabbing...")

try:
    while cam_array.IsGrabbing():
        grabResult = cam_array.RetrieveResult(10000 ,pylon.TimeoutHandling_ThrowException)

        if grabResult.GrabSucceeded():
            
            img.AttachGrabResultBuffer(grabResult)
            cameraContextValue = grabResult.GetCameraContext()
            filename = f"cam{cam_dic[cameraContextValue]['serialNum'][-4:]}_img{cam_index[cameraContextValue]:02d}.png" 
            img.Save(pylon.ImageFileFormat_Png, filename)
             
            cam_index[cameraContextValue]+=1
            if cam_index[cameraContextValue] > 100:
                cam_array.StopGrabbing()
                print(f"Stop Grabbing.")

except:
    cam_array.StopGrabbing()
    print(f"Stop Grabbing.")

    cam_array.Close()
```

thiesmoeller, 2022-02-19:
> When we connect two cameras to an industrial computer with a USB 3.0 port without the need for any additional cards (without an extra frame grabber GigE card), will we be able to run these cameras synchronously with a software trigger?

On USB 3.0 you'll always have point-to-point communication. So you can only send the software triggers one after the other.
This will give a typical delay of 500µs to a 1ms, but could be worse, if your operating system halts your process for longer ....
Depending on your realtime requirements, this could be sufficient .... You have to check this

> Are there any disadvantages of triggering with software trigger?

Software trigger always has your host and it's operating system in the game. So a lot of components, that you don't control..

> Are there any disadvantages to using USB 3.0 instead of GigE?

GigE allows for easier synchronization ( ActionCMD and IEEE1588 ) USB3 allows for higher Bandwidth....

> Are there any disadvantages to using the computer's USB 3.0 ports instead of the USB 3.0 Interface Card?

There is absolutely no difference in timing ( both are PCIe devices ) But computers USB3 ports on e.g. Intel have higher accumulated bandwidth, than an external multiport card.

> In other words, will it be technically possible for us to provide simultaneous software triggers, take pictures of two cameras at the same time, and transmit the pictures they took to the computer simultaneously?

On USB3.0: If you mean with "simultaneously" _no_ timing difference between the devices, the answer is no.
This is only reachable with a HW trigger.

For HW Trigger, you can also use one of the cameras as "main-camera" and trigger the others from the first camera`s timing.
You just have to wire up the IO-Ports of the cameras
Or use the UserOutput of one camera to trigger itself and the other ones.

FlorentLM, 2022-07-05:
Hi,

Sorry to hijack this issue.

I have a somewhat dumb question on the subject as I have very bad electrical knowledge:
@thiesmoeller Would you have an example schematic for simultaneously triggering multiple cameras with, say an Arduino or a RPi? Can you just branch all cameras off the same 5V line? Or do you *have* to have a main camera and the rest as slaves?

jhacsonmeza, 2022-07-05:
Hi @FlorentLM
You need to use the signal from your Arduino/RPi as input to all your cameras. This means that all your cameras must be connected to the same 5V output line from your Arduino and also to the same ground. You don't have a kind of main camera, because the signal is going to trigger all the cameras simultaneously. Off course, you need to enable the external trigger option in all your cameras for it to work. I don't know if you will do this in code or with the GUI.

thiesmoeller, 2022-07-06:
@FlorentLM and @jhacsonmeza 
check the manual for details:
e.g. on a Basler ace USB:
schematics for trigger circuit: https://docs.baslerweb.com/circuit-diagrams-%28ace%29.html#gpio-input-line
electrical spec: https://docs.baslerweb.com/aca2000-165um.html#operation-as-input

Regarding multiple cameras: check the max sink current of your of your arduino/rpi with the number of cameras you want to connect.
e.g. Basler ace USB specifies a max sink current of 2mA.
if your arduino/rpi can e.g. has a limit of 16mA you could trigger up to eight cameras from a single gpio.
If you have to trigger more cameras , you have to use an  external buffer to support the higher load.

FlorentLM, 2022-07-06:
@thiesmoeller, @jhacsonmeza - Thanks both, it works :)

aruna-ram, 2022-07-07:
@thiesmoeller
I'm trying to frame synchronise two cameras.
I'm generating a square wave signal to one raspberry pi gpio pin. This will be the FrameStart trigger. My current issue is sending this wave to two gpio pins *simultaneously*, one pin for each camera. I haven't found a way to do this (from current searches and basic electronics knowledge...). 
> On USB3.0: If you mean with "simultaneously" _no_ timing difference between the devices, the answer is no. This is only reachable with a HW trigger.
>
> For HW Trigger, you can also use one of the [...]

In your last sentence, do you mean that we could connect the Arduino to one camera (Camera 1) and then set its User Output feature to follow the trigger signal and transmit this to the next camera, like in the attached figure. This would solve the problem of generating the same signal on multiple GPIO pins. Would this method be synchronous - i.e. no timing difference between devices? 
[image]

Thanks in advance!

thiesmoeller, 2022-07-07:
You can just connect the same arduino pin to both cameras in parallel to both cameras.

On a standard arduino uno the GPIO can source and sink 20mA ( https://docs.arduino.cc/hardware/uno-rev3 )
So you would be able to connect up to 10 cameras directly ( we sink 2mA )

aruna-ram, 2022-07-08:
@jhacsonmeza Yes it really does, thank you very much!

thiesmoeller, 2022-07-08:
and in schematic form:
[image]

tanmoysil, 2025-03-06:
Hi @thiesmoeller 
Sorry for hijacking this thread. I have a similar question regarding triggering a Basler ace 2 camera.

I'm planning to use a USB TTL device instead of an Arduino for triggering. My oscilloscope shows the TTL device outputs around 123 mA, but I noticed the camera's opto-coupled input port has a specification of "<15 mA internally limited" and requires ">5 mA to drive the opto-coupler."

Since the TTL output current is about 8x higher than the camera's specified limit, I'm concerned about potential issues. Would you advise adding a current limiting resistor when connecting these devices?

Thank you for your guidance!

thiesmoeller, 2025-03-12:
The camera has the current limiter included (~ max 15mA ) so just connect your output signal

cmaggi82, 2025-05-31:
Hello @thiesmoeller ,

I think I have succesfully triggered 3 basler usb3 cameras (acA2040-120um) with Arduino Uno connecting in parallel as suggested.

I can check this by recording a stopwatch and verify that the time interval between captured frames is the one expected from the trigger.

I am now having some difficulties in getting simultaneous frames from the 3 cameras with pylon (most probably due to my inexperience with pylon).

In particular I start teiggering with Arduino and then I Start Grabbing with the Camera Array and retrive results in a loop.

I was (naively) expecting that 3 frames are taken from cam1, cam2 and cam3 consecutlvely at each step.

Unfortunately it happens sometimes (e.g. a few times in about 1000 frames) that that 2 consecutive frames are taken from the same camera (say cam2).

 Could you provide any example code to avoid this and always take the simultaneous images from the 3 individual cameras ?

Thank you in advance!

thiesmoeller, 2025-05-31:
Please create a new ticket. 

Can you post your code so we can help better ?

cmaggi82, 2025-06-01:
> Please create a new ticket.
> 
> Can you post your code so we can help better ?

New issue created (https://github.com/basler/pypylon/issues/842) and code shared! Thank you in advance!