basler/pypylon #475: Illegal instruction error upon import

Terizian, 2022-02-28:
I've been trying to install pypylon from source on Ubuntu 18.04 with an ARM processor. 

I followed the steps of the README file:
- Compiler: gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
- Python development files: python-dev/bionic,now 2.7.15~rc1-1 arm64 [installed]
- Swig: SWIG Version 4.1.0
-pylon from the debian package available on the basler website

I have a virtualenv set up for my python environment and I'm using 3.6.9
When I try to import the library to try some of the samples, I get the follow error message: 
`>>> from pypylon import pylon
Illegal instruction (core dumped)
`

SMA2016a, 2022-03-04:
would you try run you app with sudo?

Terizian, 2022-03-05:
It turned out to be caused by the numpy library installed. I had to downgrade to 1.19.3 for it to work.

SMA2016a, 2022-03-05:
@Terizian  : Thanks for the update, then change the status to closed.

closed 2022-03-05 by Terizian

thiesmoeller, 2022-03-08:
For reference:

This is the numpy openblas cpu detection issue https://github.com/numpy/numpy/issues/18131

and can be fixed by either up/down grading or setting the environment variable `OPENBLAS_CORETYPE=ARMV8`