basler/pypylon #477: _genicam.OutOfRangeException: The difference between Value = 450 and Min = 32 must be dividable without rest by Inc = 32. : OutOfRangeException thrown in node 'Width' while calling 'Width.SetValue()' (file 'integert.h', line 79)

DimiQI, 2022-03-07:
I work with acA2500-60uc camera.
When i set Image ROI:
#Image ROI Configuration
camera.Width.SetValue(1080)
camera.Height.SetValue(480)
camera.OffsetX.SetValue(32)
camera.OffsetY.SetValue(32)
There will be an error:
Traceback (most recent call last):
  File "D:/Graduation Design/GrabAFrame.py", line 14, in <module>
    camera.Width.SetValue(450)
  File "D:\Graduation Design\venv\lib\site-packages\pypylon\genicam.py", line 2159, in SetValue
    return _genicam.IInteger_SetValue(self, Value, Verify)
_genicam.OutOfRangeException: The difference between Value = 450 and Min = 32 must be dividable without rest by Inc = 32. : OutOfRangeException thrown in node 'Width' while calling 'Width.SetValue()' (file 'integert.h', line 79)

I wonder if the value i set must be dividable without rest by Inc = 32 or i missed something?

SMA2016a, 2022-03-07:
yes, some camera sensor has increment. 

There is a function to readout  given increment

camera.Height.GetIInc()   something like this.

DimiQI, 2022-03-09:
ok. I will try it. Thanks a lot.