basler/pypylon #478: Is there a waitkey function in pylon? How to stream video before you save video.

gt8mar, 2022-03-16:
Is there a way to stream video from a camera, then have a waitkey that causes the camera to begin saving a series of images?

I'm imagining setting a key like "r" to mean record. Then streaming video without saving until I press "r", at which point it begins saving. I tried using input() but that stops the loop where I stream the video.

thiesmoeller, 2022-03-17:
What do you mean by "stream video"?

Interacting with the keyboard is not part of pypylon.

 It depends on your type of application ( graphical/command line) and the operating system how to best register a key press.

gt8mar, 2022-03-17:
By stream I mean grab and then display video without saving. I want to replicate the pylon viewer "continuous shot" and "record" buttons. I would use the pylon viewer gui but you can't save video from both cameras at the same time.

SMA2016a, 2022-03-18:
You need to run pylon viewer in 2 instances, or you need to make your own GUI Based application.