basler/pypylon #479: No device is available

alihan-ozturk, 2022-03-23:
Hi, I have Cognex CIC-5000R camera. I want to capture image using python. I installed the necessary conditions(latest version Pypylon Camera Software Suite, latest PyPylon, swig>=4.0). I can only see and use my camera in specific pylon software. However, I am getting the following error "No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'tlfactory.cpp', line 694)". In pylon viewer my camera is not showing in the list. Only the specific version 5.0.10.10613 appears in the list and I can use it on viever. I still get the same error in python "No device is available or no ... RuntimeException thrown (file 'tlfactory.cpp', line 694)"

SMA2016a, 2022-03-26:
can you see the camera using pylon viewer? if not, most properly, you need a special license key. Contact your  local distributor.

alihan-ozturk, 2022-03-27:
Thank you for your comment. The local distributor said the same.

closed 2022-03-27 by alihan-ozturk