basler/pypylon #480: Frames missing for longer periods of time

drajsel, 2022-03-24:
Hello, 

I have a setup of 4 Basler a2a1920-51gc PRO cameras on `Ubuntu 20.04.1 LTS x86_64`. Each camera is triggered with a hardware trigger, two of them are triggered by the same trigger. The usual frame rate is about 2.2 FPS, sometimes up to 4 FPS. The network is heavily utilized (1 Gbps). `pypylon` version is `1.7.4`.

On these two cameras that are triggered at the same time I noticed that couple of times a day frames from one of the cameras stop coming entirely, that is, timeout exception occurs for around 12 minutes. It's the same camera always (with larger ROI). At the same time, a spike in the network traffic per packets occurs. It seems like the camera tries to resend failed packets and maybe gets stuck in the resend mechanism for this ~10 minutes? I came upon this scenario once and restarting the camera worked, it began receiving frames again.

I tried using the bandwidth manager, but it didn't help. Enabling jumbo packets is not currently an option for this infrastructure. Does someone have any ideas what happens and how to avoid dropping so many frames?

SMA2016a, 2022-03-26:
Then try to increase the value for interpacket delay. if possible, create Wireshark log and send it directly to me.

drajsel, 2022-03-31:
I tried increasing the interpacket delay multiple times to higher values. I also changed image format from YCbCr to Bayer RG8 to decrease the payload. However, the period of frames not coming now seems prolonged (now around 25 minutes, before ~10 minutes). I wrote a script that will try to catch the moment when the frames from one of the cameras stop coming and execute `tcpdump`, since that usually happens during the night and I can't have `tcpdump` started all the time since the file size increases rapidly. I'll send the file as soon as the situations occurs again.

I have to mention that I made a mistake earlier. The two cameras sharing the hardware trigger aren't triggered at the same time, one of them recieves the trigger 300 ms after the other.

SMA2016a, 2022-03-31:
try to use a capture filter like this.

<-i 1>  adapt you NIC ID

dumpcap.exe -f "ether[0x2e:1] < 0x01" -i 1 -w [C:\Log\basler.pcap]() -s 128 -b files:4 -b filesize:400000

drajsel, 2022-04-11:
Sorry it's taking so long, I managed to capture the traffic in the next occurence, but it seems it doesn't have all the information as the network infrastructure is more complicated than I thought. As I am an ML engineer, I asked someone from DevOps team to help me out and we'll try again. I'll get back to you as soon as I capture something significant.

drajsel, 2022-04-13:
Can I contact you via email? I can share the file with captured packets, but it may not contain all the necessary information (UDP protocol).

SMA2016a, 2022-04-15:
sure I will send you a private email.