basler/pypylon #481: Ways to Open() camera in Docker (without MacVLAN)

maximehyh, 2022-03-28:
Hello, 

I am trying to deploy a Docker container in a swarm cluster on a **Windows host** in charge of connecting and getting a stream from a Basler camera. 

We previously never had an issue since we were deploying this type of container on a **Linux host** which allows us to use MacVLANs and connect to the camera without issues since it was just as if the container was on the same network as the host. Unfortunatly MacVLANs are not supported on Windows (https://docs.docker.com/network/network-tutorial-macvlan/). Note that it is also working fine on `docker-compose` and `docker` when using `network: host`. Unfortunately we need to use Swarm for robustness matters. 

As the usual `devices = tlFactory.EnumerateDevices()` (https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py) will not work in such network architecture, we started to use the method described [HERE](https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/how-can-i-create-a-gige-camera-device-object-with-a-known-ip-address/14991/) to connect to the camera using the (static) IP of the camera. 

This is working fine and we can actually see the camera and get information such as the id. 

Unfortunately, thinks are starting to get trickier when we try to `Open()` the camera and grab a stream from it. 

Looking at the documentation, GigE spec, and Wireshark logs it seems like some UDP ports are used for a "pre connection" that will then allow the Stream Grabbing process to begin. Knowing that we then started to look for a way in the Pylon SDK to set those UDP ports and expose them on Docker. We find and successfully used the `StreamGrabber.DestinationPort` method but this seems to be only for the Stream Grabbing part and not for the Control/pre connection part. This preliminary phase seems to use multiple random ports and we cannot manage to see how to set them. 

**Question 1**: would it be possible to set the port for this pre grabbing/connection process? I am thinking that maybe exposing those/this port on docker could fix the issue.

**Question 2**: I am seeing that some TCP connection is starting and failing when camera is without network: host. Maybe some failling step at a lower network level could cause the TCP connection to be tried (as the UDP one failed)?

**Question 3**: More generally would you recommend a way to connect to a camera using IP that is not on the same network as the SDK? 

I hope this is clear enough. below are some wireshark screenshot and code snippets that could maybe help you understand our issue.

```cpp
doConnection(const std::vector<std::string> &ip_list)
{
	// Cf https://www.baslerweb.com/en/sales-support/knowledge-base/frequently-asked-questions/how-can-i-create-a-gige-camera-device-object-with-a-known-ip-address/14991/
	CTlFactory& TIFactory = CTlFactory::GetInstance();
	ITransportLayer *pTI = TIFactory.CreateTl( Pylon::CBaslerGigEInstantCamera::DeviceClass() );

	try {
		for(int i = 0; i < ip_list.size(); ++i) {
			CBaslerGigEDeviceInfo bdi( pTI->CreateDeviceInfo() );
			bdi.SetIpAddress(ip_list[i].c_str());

			Pylon::CBaslerGigEInstantCamera ip_camera( pTI->CreateDevice( bdi ) );

                        ip_camera.Open();
		}
	} catch (GenICam_3_1_Basler_pylon_v5_1::RuntimeException &e) {
		std::ostringstream oss;
		oss << "[BaslerCamera] - " << e.what();
		throw std::runtime_error(oss.str());
	} catch (GenICam_3_1_Basler_pylon_v5_1::LogicalErrorException &e) {
		std::ostringstream oss;
		oss << "[BaslerCamera] - " << e.what();
		throw std::runtime_error(oss.str());
	} catch (GenICam_3_1_Basler_pylon_v5_1::AccessException &e) {
		std::ostringstream oss;
		oss << "[BaslerCamera] - " << e.what();
		throw std::runtime_error(oss.str());
	}
}
```

Result with `network: host`:

[image]

Without `network: host`:
```bash

# Error when trying to Open()
terminate called after throwing an instance of 'std::runtime_error'
  what():  [BaslerCameraContinuous] - No device attached. : RuntimeException thrown (file 'InstantCameraImpl.h', line 2077)
```

[image]

Thanks!

SMA2016a, 2022-03-29:
<Question 3: More generally would you recommend a way to connect to a camera using IP that is not on the same network as the SDK?> 

first you need to make sure that UDP routing is working to the camera.

if the UDP routing to the camera is working, then you can Announce the remote IP before calling AnnounceRemoteDevice();

then SDK will see the camera.

or 

Pylon::CBaslerGigEInstantCamera ip_camera( pTI->CreateDevice( bdi ) );  // this method also open a camera that is not in same network, when routing is working

maximehyh, 2022-03-30:
@thiesmoeller at the moment the container is a docker container running on Docker Linux. We will try to deploy on Windows once we figure out a way to deploy the container without using a MacVLAN (which is equivalent to run docker on any host without `network: host`, i.e having container and host on the same network). I'll build something that allows you to easily test this. 

@SMA2016a 
> first you need to make sure that UDP routing is working to the camera.
That is exactly the issue, in Docker and unless you set `network: host` or use a MacVLAN, you need to specify which UDP ports to expose. I can expose the one that I manually set for the `StreamGrabbing` part but I can not expose the ones that are randomly used for the controlling part before the `StreamGrabbing` part begins.

SMA2016a, 2022-03-30:
As you mentioned, the streaming port can be set but control channel.

maximehyh, 2022-03-30:
Yes but as mentionned before I am looking for a solution where I can make this works without a MACVLAN as MacVLANs are not handled on Windows (https://docs.docker.com/network/network-tutorial-macvlan/)

maximehyh, 2022-04-01:
One way of understanding if this is feasible would be to know if you know of cases where the camera and the Pylon SDK are on two different networks?

As for example if the camera was exposing an IP somewhere on the internet (security matters set aside) and the SDK was connecting and grabbing some content from another network where firewall rules were applying. How would/could such a connection be made? 

Also have you ever encountered cases where the Pylon SDK was in a container in a Kubernetes cluster? 

Thanks!

SMA2016a, 2022-04-01:
sure you can also access the camera via internet.

you need to make sure Home router 1 forwarded all UDP packets to camera.

PC with pylon SDK<-->Home Router 2<-->Internet (DynDNS><-->Home Router 1<--> camera

maximehyh, 2022-04-02:
@SMA2016a thanks a lot. That brings us to the original question, would it be possible to manually set all the necessary UDP ports? (other than the `StreamGrabbing` one) :)

SMA2016a, 2022-04-03:
no, The port for the  control channel can not be defined on host side. But the port on camera is a fixed one.  Try to forward 3956  on "Home Router 1"  to camera.

maximehyh, 2022-04-19:
FYI the only way I found to all connecting to the camera is by setting up a MacVLAN (using Multus on Kubernetes).