basler/pypylon #484: Hardware triggering does not work in code

vitalijsfescenko, 2022-03-30:
I have issue with hardware triggering (from Arduino Uno with 120Hz signal). I have 2 Basler daA1920-160uc USB cameras each with configuration for hardware triggering from Line1 stored in "UserSet1", which works fine with hardware trigger when loaded in Pylon Viewer, however when I run code below it automatically turns off TriggerMode value and runs in Continuous Acquisition mode ignoring hardware trigger frequency and even works without it. This code worked fine in Software trigger mode.  Setting TriggerMode value to "On" from code does not change anything despite GetValue returning "On". After running code PylonViewer also shows this value as disabled, and when manually turned on it does enable HW triggering in PylonViewer, so other parameters are likely not affected or responsible. Disconnecting and reconnecting cameras in PylonViewer does not alter config so I think only code is responsible. I also tried other code for single camera similar to https://github.com/basler/pypylon/blob/master/samples/grab.py, but same behavior can be observed.

Code below is using event handlers for grabbing images from multiple cameras and is combination of https://github.com/basler/pypylon/blob/master/samples/grabcameraevents.py and  https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py.

```
from pypylon import genicam
from pypylon import pylon

import time

maxCamerasToUse = 2

class ImageEventPrinter(pylon.ImageEventHandler):
    def OnImagesSkipped(self, camera, countOfSkippedImages):
        print("OnImagesSkipped event for device ", camera.GetDeviceInfo().GetModelName())
        print(countOfSkippedImages, " images have been skipped.")
        print()

    def OnImageGrabbed(self, camera, grabResult):
        print("OnImageGrabbed event for device ", camera.GetDeviceInfo().GetModelName())

        # Image grabbed successfully?
        if grabResult.GrabSucceeded():

            # When the cameras in the array are created the camera context value
            # is set to the index of the camera in the array.
            # The camera context is a user settable value.
            # This value is attached to each grab result and can be used
            # to determine the camera that produced the grab result.
            cameraContextValue = grabResult.GetCameraContext()

            # Print the index and the model name of the camera.
            print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetModelName())

            print("SizeX: ", grabResult.GetWidth())
            print("SizeY: ", grabResult.GetHeight())

            img = grabResult.GetArray()
            print("Gray values of first row: ", img[0])
            print()
        else:
            print("Error: ", grabResult.GetErrorCode(), grabResult.GetErrorDescription())

try:
    # Get the transport layer factory.
    _TlFactory = pylon.TlFactory.GetInstance()

    # get all available cameras
    lstDevices = _TlFactory.EnumerateDevices()

    if len(lstDevices) == 0:
        print("\n   No Basler camera devices are connected!")
        sys.exit(1)

    def print_cameras(lstDevices):
        txt = ""
        for i, dev in enumerate(lstDevices):
            txt += "\n  "+str(i)+". "+dev.GetFullName()+" "+dev.GetFriendlyName()
        return txt

    print("Working camera ports are :\n"+print_cameras(lstDevices)+"\n\n")

    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    cameras = pylon.InstantCameraArray(min(len(lstDevices), maxCamerasToUse))

    for i, cam in enumerate(cameras):
        cam.Attach(_TlFactory.CreateDevice(lstDevices[i]))

        # The image event printer serves as sample image processing.
        # When using the grab loop thread provided by the Instant Camera object, an image event handler processing the grab
        # results must be created and registered.
        cam.RegisterImageEventHandler(ImageEventPrinter(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)

        # Load saved config
        cam.Open()
        cam.UserSetSelector = "UserSet1"
        cam.UserSetLoad.Execute()
            

        cam.TriggerSelector.SetValue('FrameStart')
        cam.TriggerMode.SetValue('On') #hardware trigger
        cam.TriggerSource.SetValue('Line1')
        cam.TriggerActivation.SetValue('RisingEdge')
        cam.Close()
            
        print("Using device ", cam.GetDeviceInfo().GetModelName())

    # Start the grabbing using the grab loop thread, by setting the grabLoopType parameter
    # to GrabLoop_ProvidedByInstantCamera. The grab results are delivered to the image event handlers.
    # The GrabStrategy_OneByOne default grab strategy is used.
    cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly, pylon.GrabLoop_ProvidedByInstantCamera)

    # Wait for user input to trigger the camera or exit the program.
    # The grabbing is stopped, the device is closed and destroyed automatically when the camera object goes out of scope.
    while True:
        time.sleep(1)
                
    cameras.StopGrabbing()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.", e.GetDescription())
```

thiesmoeller, 2022-03-30:
Remove the Close() in your code

vitalijsfescenko, 2022-04-01:
> […](#)

Thanks, it works now

closed 2022-04-01 by vitalijsfescenko