basler/pypylon #485: Multi-camera synchronous acquisition cannot be displayed normally

dogcat-ai, 2022-04-04:
I try to use four Basler aca-720-290gm cameras to collect and display images synchronously through PTP. Although the program can run normally, there is a problem with the display of GUI interface. When four cameras collect continuously, only some of the images of the camera may be displayed normally, or the image display interface of a camera may get stuck after a period of normal operation.

code:

```
from pypylon import pylon
from PyQt5 import QtCore,QtWidgets,QtGui
import sys
import Gui
from PyQt5.QtWidgets import QApplication, QMainWindow

# Image event handler.
class SampleImageEventHandler1(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):

        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_RGB8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        image = converter.Convert(grabResult)
        img = image.GetArray()
        x = grabResult.GetWidth()
        y = grabResult.GetHeight()
        scale = ui.camera1.height() / y
        frame = QtGui.QImage(img, img.shape[1], img.shape[0], QtGui.QImage.Format.Format_RGB888)
        pix = QtGui.QPixmap.fromImage(frame).scaled(x * scale, y * scale)
        ui.camera1.setPixmap(pix)

class SampleImageEventHandler2(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):

        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_RGB8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        image = converter.Convert(grabResult)
        img = image.GetArray()
        x = grabResult.GetWidth()
        y = grabResult.GetHeight()
        scale = ui.camera2.height() / y
        frame = QtGui.QImage(img, img.shape[1], img.shape[0], QtGui.QImage.Format.Format_RGB888)
        pix = QtGui.QPixmap.fromImage(frame).scaled(x * scale, y * scale)
        ui.camera2.setPixmap(pix)

class SampleImageEventHandler3(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):

        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_RGB8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        image = converter.Convert(grabResult)
        img = image.GetArray()
        x = grabResult.GetWidth()
        y = grabResult.GetHeight()
        scale = ui.camera3.height() / y
        frame = QtGui.QImage(img, img.shape[1], img.shape[0], QtGui.QImage.Format.Format_RGB888)
        pix = QtGui.QPixmap.fromImage(frame).scaled(x * scale, y * scale)
        ui.camera3.setPixmap(pix)

class SampleImageEventHandler4(pylon.ImageEventHandler):
    def OnImageGrabbed(self, camera, grabResult):

        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_RGB8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
        image = converter.Convert(grabResult)
        img = image.GetArray()
        x = grabResult.GetWidth()
        y = grabResult.GetHeight()
        scale = ui.camera4.height() / y
        frame = QtGui.QImage(img, img.shape[1], img.shape[0], QtGui.QImage.Format.Format_RGB888)
        pix = QtGui.QPixmap.fromImage(frame).scaled(x * scale, y * scale)
        ui.camera4.setPixmap(pix)

def Initial_cameras():
    # Get the transport layer factory.
    tlFactory = pylon.TlFactory.GetInstance()
    # Get all attached devices and exit application if no device is found.
    devices = tlFactory.EnumerateDevices()
    if len(devices) == 0:
        raise pylon.RuntimeException("No cameta present.")
    # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
    global cameras
    cameras = pylon.InstantCameraArray(min(len(devices), 4))

    for i, cam in enumerate(cameras):
        # Attach all Pylon Devices.
        cam.Attach(tlFactory.CreateDevice(devices[i]))
        cam.Open()
        # Enable PTP
        cam.GevIEEE1588.SetValue(True)
        # Make sure the frame trigger is set to Off to enable free run
        cam.TriggerSelector.SetValue('FrameStart')
        cam.TriggerMode.SetValue('Off')
        # Let the free run start immediately without a specific start time
        cam.SyncFreeRunTimerStartTimeLow.SetValue(0)
        cam.SyncFreeRunTimerStartTimeHigh.SetValue(0)
        # Set the trigger rate to 1 frames per second
        cam.SyncFreeRunTimerTriggerRateAbs.SetValue(1)
        # Apply the changes
        cam.SyncFreeRunTimerUpdate.Execute()
        # Start the synchronous free run
        cam.SyncFreeRunTimerEnable.SetValue(True)
    # create and register an image event handler processing the grab results
    cameras[0].RegisterImageEventHandler(SampleImageEventHandler1(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    cameras[1].RegisterImageEventHandler(SampleImageEventHandler2(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    cameras[2].RegisterImageEventHandler(SampleImageEventHandler3(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)
    cameras[3].RegisterImageEventHandler(SampleImageEventHandler4(), pylon.RegistrationMode_Append, pylon.Cleanup_Delete)

def start():
    cameras.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)

def stop():
    cameras.StopGrabbing()

if __name__ == '__main__':
    app = QApplication(sys.argv)
    MainWindow = QMainWindow()
    ui = Gui.Ui_MainWindow()
    ui.setupUi(MainWindow)
    ui.Connect.clicked.connect(lambda: Initial_cameras())
    ui.Start.clicked.connect(lambda: start())
    ui.Stop.clicked.connect(lambda: stop())
    MainWindow.show()
    sys.exit(app.exec_())
```

thiesmoeller, 2022-04-05:
This is most probably a multithreading issue:
pylon runs the image event handler in background threads. 
and Qt doesn't support directly calling methods of gui-objects from outside the GUI-Thread ( main-thread )

The Qt recommended way is to use the signal-slot system for a properly synchronized and queued communication.

dogcat-ai, 2022-04-06:
@thiesmoeller 
Thank you very much for your advice . I try to use the signal-slot mechanism in pyqt5 to realize the synchronous acquisition and display process of four cameras. The images collected by the four cameras can be displayed on the GUI interface. However, when four cameras continuously collect images, the image refresh process in the GUI interface is not synchronized.

code:

```
from pypylon import pylon
from PyQt5 import QtGui
from PyQt5.QtGui import QImage, QPixmap
from PyQt5.QtCore import QThread, pyqtSignal
import sys
import Gui
from PyQt5.QtWidgets import QApplication, QMainWindow

# Image event handler.
global converter
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_RGB8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

class Show_Camera1(QThread):
    show_camera1_signal = pyqtSignal(QImage)
    def run(self):
        while cameras[0].IsGrabbing():
            grabResult = cameras[0].RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                image = converter.Convert(grabResult)
                img = image.GetArray()
                x = img.shape[1]
                y = img.shape[0]
                scale = mainwindow.camera1.height() / y
                frame = QtGui.QImage(img, x, y, QtGui.QImage.Format.Format_RGB888).scaled(x * scale, y * scale)
                self.show_camera1_signal.emit(QImage(frame))
                grabResult.Release()

class Show_Camera2(QThread):
    show_camera2_signal = pyqtSignal(QImage)
    def run(self):
        while cameras[1].IsGrabbing():
            grabResult = cameras[1].RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                image = converter.Convert(grabResult)
                img = image.GetArray()
                x = img.shape[1]
                y = img.shape[0]
                scale = mainwindow.camera2.height() / y
                frame = QtGui.QImage(img, x, y, QtGui.QImage.Format.Format_RGB888).scaled(x * scale, y * scale)
                self.show_camera2_signal.emit(QImage(frame))
                grabResult.Release()

class Show_Camera3(QThread):
    show_camera3_signal = pyqtSignal(QImage)
    def run(self):
        while cameras[2].IsGrabbing():
            grabResult = cameras[2].RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
            if grabResult.GrabSucceeded():
                image = converter.Convert(grabResult)
                img = image.GetArray()
                x = img.shape[1]
                y = img.shape[0]
                scale = mainwindow.camera3.height() / y
                frame = QtGui.QImage(img, x, y, QtGui.QImage.Format.Format_RGB888).scaled(x * scale, y * scale)
                self.show_camera3_signal.emit(QImage(frame))
                grabResult.Release()

class Show_Camera4(QThread):
    show_camera4_signal = pyqtSignal(QImage)
    def run(self):
        while cameras[3].IsGrabbing():
            grabResult = cameras[3].RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
[... 61 lines cut ...]
            cam.SyncFreeRunTimerTriggerRateAbs.SetValue(2)
            # Apply the changes
            cam.SyncFreeRunTimerUpdate.Execute()
            # Start the synchronous free run
            cam.SyncFreeRunTimerEnable.SetValue(True)
            # cam.StartGrabbing(pylon.GrabStrategy_OneByOne)
            print('cam initial succeed')

    def start(self):
        cameras.StartGrabbing(pylon.GrabStrategy_OneByOne)
        self.show1.start()
        self.show2.start()
        self.show3.start()
        self.show4.start()

    def stop(self):
        cameras.StopGrabbing()

if __name__ == '__main__':
    app = QApplication(sys.argv)
    mainwindow = MyWindow()
    mainwindow.show()
    sys.exit(app.exec_())

```

thiesmoeller, 2022-04-06:
... for gui display application, where you as a human watch the result, I would recommend to show only the latest images.

Otherwise you can create a queue longer than one frame, if the display of one of the cameras delayed...
so using `cameras.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)` would realize this.

Especially with the PTP setup you are using, you should first poll the devices, if they are properly in sync ( https://docs.baslerweb.com/precision-time-protocol#checking-the-status-of-the-ptp-clock-synchronization )

To debug, if the images are in sync, I would recommend to also transfer the TimeStamp of each image to your gui. They should be in sync within the sync precisision of your PTP configuration.

and to the Qt implementation.... ( haven't tested it )
I think you have to decorate the slots like here:

```python
    @PyQt5.QtCore.pyqtSlot(QImage)
    def image_callback1(self, image):
        self.camera1.setPixmap(QPixmap.fromImage(image))
```