basler/pypylon #488: Issue while getting chunk data, "Node not existing", even though I have created it.

jokubas11, 2022-04-06:
Hi,

The camera I am using is acA2040-90umNIR.

I have been following the example to get additional chunk data from here: https://github.com/basler/pypylon/blob/master/samples/grabchunkimage.py

The example says that the GenICam nodemap is required, and it because it is time consuming, we create static pool of nodemaps beforehand:
```
self.camera.StaticChunkNodeMapPoolSize = self.camera.MaxNumBuffer.GetValue()
```
Which works well...

After that I try to enable the chunk mode:
```
if genicam.IsWritable(self.camera.ChunkModeActive):
    self.camera.ChunkModeActive = True
```
upon which I receive the following error:
```
Node not existing (file 'genicam_wrap.cpp', line 16130)
```
although it seemed that the static pool of nodes has been created succesfully, it still says that node is not existing.
Also note that the following line returns true:
```
print(genicam.IsWritable(self.camera.ChunkModeActive))
```
so it appears that ChunkModeActive is writable, but I still cannot write it and get an error...
What could be the issue here?

thiesmoeller, 2022-04-07:
can you post:

```python
import pypylon.pylon as py
import pypylon.genicam as geni

tlf = py.TlFactory.GetInstance()

cam = py.InstantCamera(tlf.CreateFirstDevice())

cam.Open()

print(cam.DeviceModelName.Value)
print(cam.DeviceFirmwareVersion.Value)

print(cam.ChunkSelector.Symbolics)
print(cam.ChunkModeActive.Value)

```

jokubas11, 2022-04-07:
acA2040-90umNIR
106555-29;U;acA2040_90u;V4.1-1;0
('Gain', 'ExposureTime', 'Timestamp', 'LineStatusAll', 'CounterValue', 'SequencerSetActive', 'PayloadCRC16')
False

SMA2016a, 2022-04-16:
Now simply try to extend this code to following. Most properly it will work.

if genicam.IsWritable(cam.ChunkModeActive):
    cam.ChunkModeActive = True
print(cam.ChunkModeActive.Value)