basler/pypylon #489: Problem: pip install pypylon

FScheiba, 2022-04-12:
can not use 
>>pip install pypylon<<

pip -V is "pip 20.0.2"
python3 -V is "Python 3.8.10"

System is Ubuntu 20.04 
What could be the problem? 
Thanks

thiesmoeller, 2022-04-12:
As we are a binary module... What is your platform ?

thiesmoeller, 2022-04-13:
And upgrade your pip version as mentioned in readme.md

`python -m pip install pip --upgrade`

matt-VR, 2023-07-02:
Same problem here.
Ubuntu 18.04 on Jetson Nano (ARM 64)
upgrading pip did not solve it. Does anybody have a solution to this?

Mert-Visea, 2023-10-27:
Did you find a solution?

ThanaphonLeonardi, 2023-11-22:
Same issue...

riccifab, 2024-02-26:
Same issue

thiesmoeller, 2024-02-26:
Python 3.12 support will come short time

efine-vivodyne, 2024-05-20:
Any update on python 3.12 support?

danielkonecny, 2024-05-28:
I'm also interested in the Python 3.12 support.

AndreasLR, 2024-05-29:
Also eager for 3.12 support