basler/pypylon #491: MemoryLocation

amarnathreddy0201, 2022-04-14:
Hi Sir/Mam,
I am getting error at   ***cameras.startgrabbing *** using cpp

```Unhandled exception at 0x00007FF811B54F69 in cppStart.exe: Microsoft C++ exception: GenICam_3_1_Basler_pylon::RuntimeException at memory location 0x000000C7D16FF280.```

How to resolve this error.

Thanks,

ANR S

SMA2016a, 2022-04-14:
it is difficult to say without having the source code.  Does it happen all the time ? or after restarting the app multiple times?

Would you share minimum buildable source code?

amarnathreddy0201, 2022-06-30:
I got it sir this is because of CMakeLists.txt and I cleared.

Thanks,

ANR .S

closed 2022-06-30 by amarnathreddy0201

Raminder2799, 2023-06-08:
Hi I'm receiving the same or

> I got it sir this is because of CMakeLists.txt and I cleared.
> 
> Thanks,
> 
> ANR .S

I'm receiving the same problem. can you explain how you solved it because I'm stuck. I'm using visual studio 2019

amarnathreddy0201, 2023-06-11:
You gonna download pypylon software of basler and inside cmake you gonna add all dlls location.

thiesmoeller, 2023-06-11:
This is the python bindings for pylon  repo.

To get everything you need for c++ development just look in the samples folder in any pylon SDK install

Raminder2799, 2023-06-24:
can you help me with this. I'm quite new and literally all the tutorials I saw confused me more. My dll files are at this location:
C:\Program Files\Basler\pylon 6\Runtime\Win32
what should I add to the basler cmake.txt?

Raminder2799, 2023-06-24:
cmake_minimum_required(VERSION 3.14)
cmake_policy(SET CMP0074 NEW)    # respect <PACKAGE>_ROOT variables in "find_package"
include(CMakePrintHelpers)

project(PylonSamples VERSION 0.0)
find_package(pylon)

add_executable( BaslerCameraSample
  BaslerCameraSample.cpp
)
target_link_libraries( BaslerCameraSample PRIVATE pylon::pylon )
install( TARGETS BaslerCameraSample )

add_custom_command(
    TARGET BaslerCameraSample POST_BUILD
    COMMAND ${CMAKE_COMMAND} -E copy_directory
        "C:/Program Files/Basler/pylon 6/Runtime/Win32"
        $<TARGET_FILE_DIR:BaslerCameraSample>
)

add_custom_target(CopyDLLs ALL DEPENDS BaslerCameraSample)
this is what I've done

amarnathreddy0201, 2023-06-25:
If  you mind please try this https://git.ligo.org/cds/software/cameras/pylon-camera-server/-/blob/main/CMakeLists.txt

amarnathreddy0201, 2023-06-25:
https://gitlab.ciirc.cvut.cz/cmirajak/laser_plane_scanner/blob/91ff49484f3d4931a8d8620343f8b4381e031226/cmake_modules/FindPylon.cmake

try this one also.

amarnathreddy0201, 2023-06-25:
set(basler_runtime "C:/Program Files/Basler/pylon 6/Runtime/x64") 
target_include_directories(simple_manager PRIVATE ${basler_runtime})

set(baslerdll "C:/Program Files/Basler/pylon 6/Development/lib/x64")
target_link_directories(simple_manager PRIVATE ${baslerdll})

amarnathreddy0201, 2023-06-25:
This is the method i used for Connecting libraries.

#Find package for Pylon
set(baslerdll "C:/Program Files/Basler/pylon 6/Development/lib/x64")
target_link_directories(simple_manager PRIVATE ${baslerdll})

set(basler_runtime "C:/Program Files/Basler/pylon 6/Runtime/x64") 
target_include_directories(simple_manager PRIVATE ${basler_runtime})

set(basler "C:/Program Files/Basler/pylon 6/Development/include")
target_include_directories(simple_manager PRIVATE ${basler})

amarnathreddy0201, 2023-06-25:
This is the best solution I used Connect this properly with directions and some directories.

https://github.com/lagadic/visp/blob/master/cmake/FindPylon.cmake