basler/pypylon #494: Difference between CInstantCameraArray vs CInstantCamera

amarnathreddy0201, 2022-05-13:
Hi Sir/Mam,

I got one doubt regarding Pylon camera accessing(CInstantCameraArray or CInstantCamera).

1) which one best for accessing the camera. either CInstantCameraArray or CInstantCamera.

2)What is the difference between this CInstantCameraArray vs CInstantCamera.

Anybody Suggest me to which one is best for accessing Cameras.

Thanks,
ANR S

SMA2016a, 2022-05-13:
CInstantCameraArray   just a array of CInstantCamera.   You can manage start and stop image acquisition with a single command.

Please refer the multi-camera sample in pylon's C++ sample collection.

amarnathreddy0201, 2022-05-13:
I got it.

CInstantCameraArray in this we have multiple cameras.

CInstantCamera  single camera

like CInstantCameraArray  we will attach the the cameras in  CInstantCamera.

closed 2022-05-13 by amarnathreddy0201