basler/pypylon #496: Is it possible to crop the image before using converter?

maxwxzheng, 2022-05-20:
I'm using async image grabbing:
`self.camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)`
In the `OnImageGrabbed` function, I'm using a converter to convert the grab result.
For my application, I need to crop the image. To improve performance, can I crop the grab result before using the converter to do the conversion?

thiesmoeller, 2022-05-20:
The features Width Height OffsetX and OffsetY control the crop ( aka RegionOfInterest )

maxwxzheng, 2022-05-20:
Thanks @thiesmoeller. I've explored both ROI and Multi-ROI features. The type of cropping I'm trying to do cannot be done with either of those features. Is there any way for me to implement my own cropping directly on the grab result?

thiesmoeller, 2022-05-20:
Can you explain the cropping you need?

maxwxzheng, 2022-05-20:
I'm trying to get the 4 smaller rectangles as shown in this image:
[image]

thiesmoeller, 2022-05-23:
In this case I would really recommend:
* use cropping on camera for the bounding box of your requested ROIs
* use numpy slicing to get the ROIs
  e.g. 
```python
...
res = cam.RetrieveResult(1000)
img = res.Array
img_0 = img[top_0:top_0+height_0, left_0:left_0+width_0]

```
* use cv2 cvtColor for the conversion ( the arguments depend on the pixelformat of your camera, but nearly all camera formats are supported in cv2 cvtColor )

thiesmoeller, 2022-05-24:
opencv uses different definition for bayer location

this is an example of a converter for the grabresult using opencv

```python
# converter class that is a drop-in replacement for the pylon converter used in many examples
class CV2_BGRConverter(object):
    # bayer conversion ( !! format definition is different in pylon and opencv !!)
    # https://docs.opencv.org/4.5.1/de/d25/imgproc_color_conversions.html
    conversion = {
        py.PixelType_BayerRG8:    lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_BG2BGR),
        py.PixelType_BayerRG12:   lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_BG2BGR),
        py.PixelType_BayerRG12p:  lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_BG2BGR),
        
        py.PixelType_BayerBG8:    lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_RG2BGR),
        py.PixelType_BayerBG12:   lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_RG2BGR),
        py.PixelType_BayerBG12p:  lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_RG2BGR),
        
        py.PixelType_BayerGR8:    lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GB2BGR),
        py.PixelType_BayerGR12:   lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GB2BGR),
        py.PixelType_BayerGR12p:  lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GB2BGR),
        
        py.PixelType_BayerGB8:    lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GR2BGR),
        py.PixelType_BayerGB12:   lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GR2BGR),
        py.PixelType_BayerGB12p:  lambda img: cv2.cvtColor(img, cv2.COLOR_BAYER_GR2BGR),
        py.PixelType_RGB8packed:  lambda img: cv2.cvtColor(img, cv2.COLOR_RGB2BGR),
        py.PixelType_BGR8packed:  lambda img: np.copy(img),
        py.PixelType_YUV422packed: lambda img: cv2.cvtColor(img, cv2.COLOR_YUV2BGR_UYVY),
        py.PixelType_YUV422_YUYV_Packed: lambda img: cv2.cvtColor(img, cv2.COLOR_YUV2BGR_YUYV),
    }

    def __init__(self):
        pass
        
    def convert(self, res : py.GrabResult):
        assert res.PixelType in self.conversion
        with res.GetArrayZeroCopy() as img:
            res_img = self.conversion[res.PixelType](img)
        return res_img
```

maxwxzheng, 2022-05-24:
Thanks @thiesmoeller !

closed 2022-05-24 by maxwxzheng

Lucrezia-Cester, 2022-08-24:
Do the ROI width height features crop before or after acquisition?