basler/pypylon #497: 5 or more cameras connect issue

MehmetOKUYAR, 2022-05-23:
[image]

I get this error when I try to use 5 or more usb basler cameras

`_genicam.RuntimeException: WaitObject duplicate failed (0): Too many open files. Reached open files limit: To change the limit permanently on a ubuntu system, edit /etc/security/limits.conf : RuntimeException thrown (file 'WaitObject_Unix.cpp', line 76)
`

and also ı get this error 

`_genicam.RuntimeException: PrepareGrab (StartStreaming) failed for device '2676:ba02:4:4:5'. Error: 'Insufficient system resources exist to complete the API.' : RuntimeException thrown (file 'PylonUsbStream.cpp', line 275)
`

how can ı solve ? is there any known solution ?

SMA2016a, 2022-05-24:
Try to increase the number of file descriptor on your system

ulimit -n 4096

MehmetOKUYAR, 2022-05-24:
actually, before I tried this code, I couldn't open more than 4 cameras in the pylon Viewer interface. After trying this code, I managed to run 4 or more cameras in the pylon Viewer interface, but in the python code I wrote, I still get an error when running more than 4 cameras.

SMA2016a, 2022-05-24:
[image]

SMA2016a, 2022-05-27:
reboot your system and then remove the following lines.

window_name = f'Camera-{cameraContextValue:03}'
            cv2.imshow(window_name, img)
            k = cv2.waitKey(1)
            if k == 27:
                break

then try to run your code, and let me know, if you can use 5 cameras without any issue

MehmetOKUYAR, 2022-05-27:
@SMA2016a Yes, ı didnt get an error like that, it worked but ı want to show images. how to is possible?

thiesmoeller, 2022-05-27:
One error in your code:
you never return the buffers to the buffer queue ( see example on readme.md ).
One way to let pypylon take care of this is the following pattern
 
```python
while cam.IsGrabbing():
    with cam.RetrieveResult(1000) as res:
        if res.GrabSucceeded():
            <do something with _res_ >

```

Here the context manager of the `with` statement will take care to release `res` in any case, when the context is left.

If you never return buffers to the grab-loop you either run out of buffers or your operating system can't handle so many open data structures ... what ever comes first

SMA2016a, 2022-05-27:
@MehmetOKUYAR  in your code, it is  missing :   grabResult.Release()

if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        cv2.namedWindow('title', cv2.WINDOW_NORMAL)
        cv2.imshow('title', img)
        k = cv2.waitKey(1)
        if k == 27:
            break
    grabResult.Release()

SMA2016a, 2022-06-02:
@MehmetOKUYAR   OK, then close this thread

MehmetOKUYAR, 2022-08-09:
> Try to increase the number of file descriptor on your system
> 
> echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb
> 
> ulimit -n 4096
> 
> let me know which one helps you

it worked with these